revarbat/BOSL #1: hex grid: beyond rectangular grids

adrianVmariano, 2019-03-08:
So far I've needed a grid structure twice, and both times not a regular uniform grid but a hex grid.  The first time I wanted it squeezed in one direction, so not uniform.  Now I need a regular one.   The existing grid module can generate a 3d grid.  But the hex grid has multiple extensions into 3d.  It's not obvious to me that handling these extensions is useful.   One question regarding a hexgrid() module is how do you define the extent of the grid.  It appears there are two basic regular cases: rectangular extent or hexagonal extent.   So for example, specify a hex grid with specified side length of the hexagonal grid and a spacing or count.  Or specify a rectangular extent by specifying spacing or a count and specifying a bounding rectangle.   In the rectangle case there's an extra complication because there are two different kinds of rows, so which kind occurs first?  I suppose the user could give a "seed" point on the grid and then the bounding rectangle.   What is the most natural way to specify a grid that is nonuniform?   Do you specify a squish factor, say on the X axis?

revarbat, 2019-03-23:
Implemented in grid2d(), in commit 993c267

closed 2019-03-23 by revarbat

revarbat, 2019-03-23:
Rather, commit 9a3c25d3d30b17b04d4bdffb39e0e4e253bf92f9