revarbat/BOSL #2: grid intersection with polygon

adrianVmariano, 2019-03-08:
Thinking about hexgrid() made me wonder if it would be useful to have a way of intersecting a grid with an arbitrary polygon.  What if I want a grid on a circle, for example.  This cannot be done after the fact because the children might overlap or have an irregular shape.  It seems like if we have a function that tests whether a grid point is inside a polygon then this should be a simple addition.  Is it a good idea?  

For hexgrid this would provide a mechanism for controlling the extent, as default cases of rectangular or hexagonal extent could be reduced to default intersection polygons.

revarbat, 2019-03-23:
Implemented in grid2d, in commit 993c267

closed 2019-03-23 by revarbat

revarbat, 2019-03-23:
Rather, commit 9a3c25d3d30b17b04d4bdffb39e0e4e253bf92f9