revarbat/BOSL #3: array_zip not proper inverse to array_subindex

adrianVmariano, 2019-03-23:
array_zip doesn't reverse the effect of array_subindex.  Also I think it would be nice if it had the option of taking 2 or 3 items to zip that aren't in a list, because I think the dominant use will be with two args, as in array_zip(a,b) rather than needing array_zip([a,b]).  

The current version zips [[1,2],[3,4],[5,6]] and [1,2,3] into [[[1,2],1],[[3,4],2],[[5,6],3]]. 

This version I think works correctly, but just bans varying lengths instead of filling in a default value.  

function array_zip(vecs,v2,v3) =
  v3!=undef ? array_zip([vecs,v2,v3]) :
  v2!=undef ? array_zip([vecs,v2]) :
  let(
    length = len(vecs[0]),
    samesize = [for (v=vecs) len(v)==length],
    dummy=assert(all(samesize),"Input vectors must have the same length")
  )
  [for(i=[0:length-1])
     [for(v=vecs) each v[i]]
  ];

revarbat, 2019-03-24:
Yeah, this got lost in all the other work.  Tracking is why I prefer issues over email.

Hmm.  I think I'd rather take python's approach.  zip() goes by shortest length list.  zip_longest() by the longest, with fill-in values.

```
function array_zip(vecs,v2,v3) =
    v3!=undef ? array_zip([vecs,v2,v3]) :
    v2!=undef ? array_zip([vecs,v2]) :
    let(minlen = min([for (v=vecs) len(v)]))
    [for(i=[0:minlen-1]) [for(v=vecs) each v[i]] ];
```

```
function array_zip_longest(vecs,v2,v3,dflt=undef) =
    v3!=undef ? array_zip_longest([vecs,v2,v3],dflt=dflt) :
    v2!=undef ? array_zip_longest([vecs,v2],dflt=dflt) :
    let(maxlen = max([for (v=vecs) len(v)]))
    [for(i=[0:maxlen-1]) [for(v=vecs) each (i<len(v)? v[i] : dflt)] ]; 
```

revarbat, 2019-03-24:
BTW, the reason I use `dflt` instead of `default` is because it avoids namespace collisions.  I honestly don't trust OpenSCAD to not get confused between the module default() and the variable default.

revarbat, 2019-03-24:
Hmm.  Actually, I want to avoid `each`, for backwards compatibility with the latest proper release, from 2015.

```
function array_zip(vecs,v2,v3) =
    v3!=undef ? array_zip([vecs,v2,v3]) :
    v2!=undef ? array_zip([vecs,v2]) :
    let(minlen = min([for (v=vecs) len(v)]))
    [for(i=[0:minlen-1]) [for(v=vecs) for (x=v[i]) x] ];
```

```
function array_zip_longest(vecs,v2,v3,dflt=undef) =
    v3!=undef ? array_zip_longest([vecs,v2,v3],dflt=dflt) :
    v2!=undef ? array_zip_longest([vecs,v2],dflt=dflt) :
    let(maxlen = max([for (v=vecs) len(v)]))
    [for(i=[0:maxlen-1]) [for(v=vecs) for (x=(i<len(v)? v[i] : dflt)) x] ];
```

adrianVmariano, 2019-03-24:
I've never noticed that OpenSCAD has trouble with mixing up functions, modules and variables.  I'd say it's confusing for *me* that these all have a separate name space.  One can make calls with parameters like "faces=faces" where the former is a function parameter and the latter a variable.  But I'd suggest that if you don't like the multiple name spaces then for a user, the default() function is sort of an internal library function, used by the library (or by library authors) and the dflt parameter name is more important to make accessible and memorable.  

I think the situation where I want to zip together items with different lengths would be rare.  (It wouldn't have occurred to me to support this at all.)  I would rather get an error than have array_zip quietly discard the end of my data.   Would it make sense to have "longest" and "shortest" be boolean flags and the default require that the lengths match?

revarbat, 2019-03-24:
`array_zip()` with assertion, `array_zip_shortest()`, and `array_zip_longest()`, renaming `dflt` to `fill`?

revarbat, 2019-03-24:
Also thinking these may be useful:
```
function array_pad(v, minlen, fill=undef) = concat(v, len(v)<minlen? [for (i=[len(v):minlen-1]) fill] : []);
function array_truncate(v, maxlen) = maxlen<1? [] : [for (i=[0:min(len(v),maxlen)-1]) v[i]];
function array_resize(v, length, fill) = array_truncate(array_pad(v,length,fill), length);
```

revarbat, 2019-03-24:
Just tested, and yes, OpenSCAD lets you have functions, variables, and modules with the same name.  This kinda makes me twitch.

adrianVmariano, 2019-03-24:
Yeah, I know you can do that and I have a similar reaction.  And parameters names sort of form a forth name space.  So you could have code that looked like thing(thing=thing(thing)){children....}
where the first one is the module, the second one is a named parameter, and then you pass in the function result of the thing() function, passing in a variable named thing.  

With regards to array_pad, array_truncate and array_resize, I wonder if we should reserve array_resize for a capability like in MATLAB of changing the dimensions of an array---though I guess you've already got array_group that somewhat does this.   (Resize in MATLAB, in case you don't know, reorganizes an array to have specified dimensions, e.g. changing a 4x6x9 array into a 2x9x12 array or a 8x27 array.)  

My reaction is that, similar to the way I can't really imagine a use for array_zip_shortest, I don't see a need for pad and truncate either.  But maybe my imagination is just limited.  It seems in particular like functions that leave the result of indeterminate length are unlikely to be that useful.  

With regards to the proposal of array_zip, array_zip_shortest and array_zip_longest I think that's OK.  I really like "fill" instead of "dflt".  I do think that it would be cleaner to just make one function that does all three things, though, with the assert as the default behavior and shortest and longest as options the user can select.  I think it's kind of name clutter to make 3 functions for this.

revarbat, 2019-03-24:
Say you had an arbitrary number of sports teams, and you wanted to pair them up to represent matches.  If you have an odd number of teams in a round, then one team gets left out that round.  Or maybe gets paired with "No-Match".

```
let (mid=floor(len(teams)/2))
array_zip_shortest(select(teams,0,mid-1),select(teams,min,-1)))
```

revarbat, 2019-03-24:
Hit wrong button.

revarbat, 2019-03-24:
Maybe call `array_truncate()` `array_trim()` instead.

revarbat, 2019-03-25:
I keep needing to pad, truncate and resize arrays all the time.  Most are covered by `point2d()` and `point3d()`.  The matrix transformation functions though, often need me to pad a 1 to the end of a point3d, so I can do a matrix x vector multiply.  I've been using concat, so far.

revarbat, 2019-03-25:
Sorry that I haven't got these checked in and fixed yet.  I've been spending all of yesterday and today building chicken coops and pens.  Too tired to do much else.

revarbat, 2019-03-25:
```
use <BOSL/compat.scad>

function array_shortest(vecs) = min([for (v = vecs) len(v)]);
function array_longest(vecs) = max([for (v = vecs) len(v)]);
function array_pad(v, minlen, fill=undef) = let(l=len(v)) [for (i=[0:max(l,minlen)-1]) i<l? v[i] : fill];
function array_trim(v, maxlen) = maxlen<1? [] : [for (i=[0:min(len(v),maxlen)-1]) v[i]];
function array_fit(v, length, fill) = let(l=len(v)) (l==length)? v : (l>length)? array_trim(v,length) : array_pad(v,length,fill);
    
function array_zip(vecs, v2, v3, fit=false, fill=undef) =
    (v3!=undef)? array_zip([vecs,v2,v3], fit=fit, fill=fill) :
    (v2!=undef)? array_zip([vecs,v2], fit=fit, fill=fill) :
    let(
        dummy1 = assert_in_list("fit", fit, [false, "short", "long"]),
        minlen = array_shortest(vecs),
        maxlen = array_longest(vecs),
        dummy2 = (fit==false)? assertion(minlen==maxlen, "Input vectors must have the same length") : 0
    )   
    (fit == "long")?
        [for(i=[0:maxlen-1]) [for(v=vecs) for (x=(i<len(v)? v[i] : fill)) x] ] :
        [for(i=[0:minlen-1]) [for(v=vecs) for (x=v[i]) x] ];
```

revarbat, 2019-03-25:
Added in [master 796fe23]

closed 2019-03-25 by revarbat

adrianVmariano, 2019-03-25:
I definitely appreciate the potential utility of both padding and truncating as methods to force a vector to have a known length.  But truncating so that a vector has length <=N, or padding so that a vector has length >=N, where the length after the operation is unknown----those things seem like weird operations. 

I have two comments on this.  One is that "fit" doesn't seem like a good keyword name.  I have tried to think of the perfect keyword and not succeeded, but I like "mode" better than "fit", which just doesn't make sense to me.   But also the options "short" and "long" aren't intuitive to me.  I would name them "pad" and "truncate" or something like that, which I think much more clearly indicates what will happen.

The other issue is that there is a better way to insert assertions into the code without dummy variables.  Are you aware of this and just doing it with dummy variables for backward compatibility, or other reasons?   If you are going to use assert properly it's impossible (as far as I know) to write a function like assert_in_list without introducing the need for clumsy dummy variables---maybe you feel the dummy variables are the lesser evil.  

Also with my version of OpenSCAD, the assertion isn't actually working at all.  It would be nice if the manual gave a specific version number when a capability was added rather than the vague "nightly build" statement.   In fact, there's kind of a problem there in that something that was added to the nightly build last year could potentially still say "nightly build" if nobody remember to update it.   I have RC3 and with the assertion() function, no message is printed.  I tried changing "assertion" to "assert" and that fixed it.  I do have the assert() function.  You test against 20190303.  My version number appears to be 20190100.   The "about" page says I have 2019.01-RC3, and the latest available is only 2019.01-RC4.  Is there a newer one hidden somewhere?