revarbat/BOSL #4: all and any should recurse into sublists

adrianVmariano, 2019-03-23:
The new functions all() and any() don't recurse down the list like the
version I supplied.  I think we want

any([[false,false],[false,false]]) to be false, not true.

and similarly

all([[true,true],[true,false],[true,true]]) should be false.

revarbat, 2019-03-25:
Recursive?  Interesting.

Original submitted code:

```
// This could be folded into listcomp()
function scalarcomp(a,b) = a<b ? -1 : a>b ? 1 : 0;

function listcomp(a,b,n=0) =   !is_list(a) && !is_list(b) ? scalarcomp(a,b):
                              len(a)<=n && len(b)<=n ? 0 :
                              len(a)<=n ? -1 :
                              len(b)<=n ? 1 :
                              a[n] < b[n] ? -1 :
                              a[n] > b[n] ? 1 :
                              listcomp(a,b,n+1);

function sort(arr) =
   !(len(arr)>0) ? [] :
       let(  pivot   = arr[floor(len(arr)/2)],
             compare = [for (entry = arr) listcomp(entry, pivot)],
             lesser  = [ for (i = [0:len(arr)-1]) if (compare[i]==-1) arr[i] ],
             equal  =  [ for (i = [0:len(arr)-1]) if (compare[i]==0) arr[i] ],
             greater = [ for (i = [0:len(arr)-1]) if (compare[i]==1) arr[i] ]
       )
       concat( sort(lesser), equal, sort(greater) );

function unique(points) = let(sorted=quicksort(points))
   [for(i=[0:len(sorted)-1]) if (i==0 || sorted[i]!=sorted[i-1]) sorted[i]];
```

revarbat, 2019-03-25:
Implemented in 3abf9fa868710459b5811020412311c6eebb45b5

closed 2019-03-25 by revarbat