revarbat/BOSL #6: better name for wrap_range

adrianVmariano, 2019-03-23:
The name wrap_range doesn't really capture the functionality.  The revised version takes various types of index sets, not just ranges.  I suggest renaming (or providing an alternate name) of select for this function.

revarbat, 2019-03-24:
I could rename it, and deprecate the old name with a pass through.  While select() isn't a bad name, I'm an old C programmer, and select() meant something faaaaaaar different.

adrianVmariano, 2019-03-24:
I was trying to think of alternatives to select() and the only thing that came to mind was index() which is arguably even more confusing for a C programmer, since it's meaning is something you could actually do.  There are no file descriptors in OpenSCAD.

revarbat, 2019-03-24:
And `select()` is already used as a name in an example in the official docs. It’s probably the best choice.

revarbat, 2019-03-25:
Deprecated `wrap_range()`.  Renamed to `select()`.  Implemented in commit [master 796fe23]

closed 2019-03-25 by revarbat