revarbat/BOSL #7: Almost all modules require functions from nightlies

sej7278, 2019-03-23:
is_list(), is_num(), is_string() and so on seem to be used everywhere. As per the [docs](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#is_string) these are only available in nightly builds, they don't even appear to be in 2019.01-RC2 as packaged for Debian Sid (afaik RC4 is the absolute bleeding edge).

I managed to get some examples to work by changing the version detection in compat.scad to look for > 20190100 (rc2) instead of 20190000.

revarbat, 2019-03-24:
Wait...  they aren't including these in the first release candidate in four years???

revarbat, 2019-03-24:
Should be fixed in [master 687f04c].  I *know* these work in 2019.03.03, and I used them in the early January dev release.  If they don't support them in the next RC after this, I will be annoyed.

sej7278, 2019-03-24:
lol, yes quite, you'd think anything from this year (or > 2015!) would include some of the more basic keywords.... thanks for the fix.

closed 2019-03-24 by sej7278

sej7278, 2019-03-24:
up() and down() don't seem to exist either.

revarbat, 2019-03-25:
Verified they do exist in BOSL/transforms.scad.
Tested using the in a short script.  They work as expected.

revarbat, 2019-03-25:
```
use <BOSL/transforms.scad>
up(20) cube(5);
```

Works.

sej7278, 2019-03-25:
ah ok, thanks, needed the extra include - i assumed they were built-ins as they seem to be in some of the unit tests in openscad with a very similar definition:

```translate([0,0,1]) child(0);```