revarbat/BOSL #11: add sort() and unique()

adrianVmariano, 2019-03-24:
Sort should do something intelligent with vectors so that unique can work on lists of points.

// This could be folded into listcomp()
function scalarcomp(a,b) = a<b ? -1 : a>b ? 1 : 0;

function listcomp(a,b,n=0) =   !is_list(a) && !is_list(b) ? scalarcomp(a,b):
                               len(a)<=n && len(b)<=n ? 0 :
                               len(a)<=n ? -1 :
                               len(b)<=n ? 1 :
                               a[n] < b[n] ? -1 :
                               a[n] > b[n] ? 1 :
                               listcomp(a,b,n+1);

function sort(arr) =
    !(len(arr)>0) ? [] :
        let(  pivot   = arr[floor(len(arr)/2)],
              compare = [for (entry = arr) listcomp(entry, pivot)],
              lesser  = [ for (i = [0:len(arr)-1]) if (compare[i]==-1) arr[i] ],
              equal  =  [ for (i = [0:len(arr)-1]) if (compare[i]==0) arr[i] ],
              greater = [ for (i = [0:len(arr)-1]) if (compare[i]==1) arr[i] ]
        )
        concat( sort(lesser), equal, sort(greater) );

function unique(points) = let(sorted=sort(points))
    [for(i=[0:len(sorted)-1]) if (i==0 || sorted[i]!=sorted[i-1]) sorted[i]];

revarbat, 2019-03-25:
I want to think for a bit on listcomp/scalarcomp.  I'll probably just put all this verbatim in a new `BOSL/sorting.scad` file.

adrianVmariano, 2019-03-26:
I'm not sure it makes sense to have a separate file just for this.  Does it make sense to add a utility.scad  or a list.scad file for list processing functions?  Some of the other stuff like array_zip and so on are not really "math" functions.  

I felt like listcomp() was kind of clumsy, so if you can think of a better solution that would be great.

adrianVmariano, 2019-03-27:
It seems like we also might want an option or separate sort function that returns the sort permutation instead of returning the sorted list.   This would enable generically the ability of "sort on nth column" for example---extract the nth column with array_subindex, sort, and use select to reorder the data.  

My current thinking is this would best be done by a separate function that creates a new data structure with the integers in order and sorts it using a comparison that ignores the extra data.

adrianVmariano, 2019-03-27:
function sort(arr,skip=0) =
        !(len(arr)>0) ? [] :
            let(  pivot   = arr[floor(len(arr)/2)],
                  compare = [for (entry = arr) listcomp(entry, pivot,skip)],
                  lesser  = [ for (i = [0:len(arr)-1]) if (compare[i]==-1) arr[i] ],
                  equal  =  [ for (i = [0:len(arr)-1]) if (compare[i]==0) arr[i] ],
                  greater = [ for (i = [0:len(arr)-1]) if (compare[i]==1) arr[i] ]
            )
            concat( sort(lesser,skip), equal, sort(greater,skip) );

     function sortind(arr) =  
        !(len(arr)>0) ? [] :  
            let(  augmented = array_zip( [for(i=[0:len(arr)-1])i], arr),
              sorted = sort(augmented, skip = 1),
           )
                array_subindex(sorted, 0);

revarbat, 2019-03-28:
Added `array_unique()` and renamed `quicksort()` to `array_sort()`.  Dev branch commit 6f5a9901243ec16e2f644db175f4a8d96518575d

adrianVmariano, 2019-03-28:
I suggest calling it sort() rather than quicksort().  The user doesn't care what algorithm is used, just that the function sorts.   If we discovered next year that the foosort() algorithm was ten times faster you'd be in the position of either having two sort commands or having a sort command called "quicksort" that implemented a different algorithm (for compatibility).

adrianVmariano, 2019-03-28:
Note also that in MATLAB I find sometimes I need to sort three variables relative to the same key list, which is when I use the feature of MATLAB to return a permutation list.  Having the subindex option allows this, but through the mechanism of zipping the array together, sorting, and then doing a series of subindex operations.  I'm not sure what use pattern is likely in OpenSCAD.  

I see you renamed it to array_sort().  What other things could possibly be sorted?  I don't see the justification for the longer name.   (BTW the two functions in MATLAB are called "sort" and "unique".)

I don't think namespace clutter is a major concern, if that's what you're worried about.  MATLAB doesn't have namespaces and has dramatically more functions than we ever will and everything works fine.

revarbat, 2019-03-28:
```
function enumerate(arr) = [for (i=[0:len(arr)-1]) [i, arr[i]]];
function sort_indexed(arr) = array_subindex(array_sort(enumerate(arr), idx=1), 0);
```

revarbat, 2019-03-28:
As for `array_sort()` vs. `sort()`, I like the array prefix not because of namespace pollution, but because it classifies the call, and associates it with other array functions thematically.  It's not a function that is used so often that the short name is particularly useful for shortening code.

To be honest, there's a number of array functions that exist now, that I'd rename if I were doing this all again.  Some like `reverse()`, I'd add the array prefix to.  Some, that are far more frequently used, I'd either keep short, or shorten based on their usage.

adrianVmariano, 2019-03-28:
What is an "array" in OpenSCAD?  To me the word suggests a list with a rectangular structure, or more formally (but recursively), a list that is either a list of scalars or a list of arrays.   As such, the sort function doesn't actually operate on arrays.   And neither does unique().    So the array prefix is questionable.  

It seems to be desirable that we choose names that extend the openscad language in a uniform and consistent way---sort of what you imagine the openscad authors would have done if they made this operation a primitive, so that the library seems like a natural extension.

adrianVmariano, 2019-03-28:
Is it worth making an index_list(N) function that generates [0,1,2,3,4,...,N-1]?  I'm not sure if this is worth doing or not.  The enumerate() function seems to me unlikely to be of general utility, but a counting vector would probably be.  (There talk of making range syntax generate lists would would make this unnecessary.)

revarbat, 2019-03-28:
I've been thinking on `array` vs. `list`, vs. `vector` recently.  These terms have been used nearly interchangeably by myself, and a lot of other docs.  However, I agree that making the definitions consistent would be good.

- List: A single-dimensional list of items.
- Vector: A list of scalars.
- Array: A multi-dimensional list-of-lists.

This would necessitate renaming a lot of things.  This may be a BOSL2 change.

revarbat, 2019-03-28:
Regarding something for returning a counting list, Python has the `range()` function for that.  Something like:

```
function range(i,j=undef,step=1) = let(
    r = (j==undef)? [0:step:i-1] : [i:step:j-1]
) [for (n=r) n];
```

The `enumerate()` function is something else I borrowed from Python.  It actually comes in handy.

revarbat, 2019-03-28:
Thinking on it I have inconsistency with "pointlist" vs "path"/"path3d" also.  The name "path" or "path3d" implies a linear ordering to the list of points, while "pointlist" would imply no ordering.

revarbat, 2019-03-28:
Given the list/array inconsistency, I admit that just `sort()` and `unique()` are fine.

revarbat, 2019-03-29:
Changed back to `sort()` and `unique()`.  Dev branch commit 326dbd47d8e82d8efd8ff25c04a76f2e792cbcff

revarbat, 2019-03-31:
Merged into master.

closed 2019-03-31 by revarbat