revarbat/BOSL #12: determining angles between vectors (or points)

adrianVmariano, 2019-03-24:
Would it be worth having a function that could determine the angle defined by 3 points in a row to avoid having to turn point sequences into vectors?  

I'm not sure I understand the existing angle functions' intent.  There is vector3d_angle (which I had a hard time remembering the name of as I tried vector_angle3d).  It works on 3d or 2d angles and uses acos of inner product.   Then there's vector2d_angle which uses difference of atan2.  

p1 = [0,0];
p2 = [1,0];
p3 = [0,-1];
echo(vector3d_angle(p1-p2, p3-p2));   // prints 45
echo(vector2d_angle(p1-p2, p3-p2));   // prints 315

It's difficult to understand when I would want the answer to be 315.   The difference of atan2 method seems subject to wraparound issues (of which this is one) that the other method doesn't have.    I guess the argument could be made that if I reversed the order, then the answer 315 would make sense, namely that the angle starting at (p3-p2) and ending at (p1-p2) is 315.  Unfortunately:

echo(vector2d_angle(p3-p2,p1-p2));   // prints -315

So it seems like perhaps vector3d_angle should be renamed vector_angle, maybe (since it's dimension independent and would in fact work in 4d) and the intent of vector2d_angle needs to be clarified and it needs to be fixed to match up with whatever that intent is.   Maybe what you want to do is run atan2 and posmod the results into [0,360] and then force the second angle to be larger than the first by adding 360 if needed, and then take the difference?   Would that always give the angle required to rotate v1 in the positive direction until it aligns with v2?  And is that the intent of this function?   (Maybe the intent is rotation in the negative direction as you subtract v2 from v1.)

revarbat, 2019-03-25:
That's definitely a bug in vector2d_angle.

revarbat, 2019-03-25:
Deprecated `vector2d_angle()` and `vector3d_angle()`.  Renamed `vector3d_angle()` to `vector_angle()`.  Implemented in 228f4c4134d1c513957093f11a10ed88184f9a66

closed 2019-03-25 by revarbat