revarbat/BOSL #13: add replist() function

adrianVmariano, 2019-03-24:
Add a function to construct lists by replicating a list to a specified length:

function replist(list, N) = [for(i=[0:N-1]) list];

If we wanted to get fancy we could make a version that would make higher depth lists, e.g. the equivalent of a 2x2 matrix of all 1's, which would just be a sequence of recursive calls.  I really don't know if arrays of higher than dimension 2 are useful or not.   Maybe it suffices to make a version that takes a third argument and makes a N x M output.

revarbat, 2019-03-25:
Not sure what you want here.  Looks like what you wrote will do:
```
replist([1,2,3], 3);  // returns [[1,2,3], [1,2,3], [1,2,3]]
```
But it sounds like you want:
```
replist([1,2,3], 3);  // returns [1,2,3,1,2,3,1,2,3]
```

adrianVmariano, 2019-03-25:
I'm using the function I wrote, so it does the right thing.  :)   I was using it both as replist(1,10) to make a length 10 vector and also replist([3,4],10) to make a length 10 x 2 array.  

My followup text was muddled, however.  What I meant was when I said higher depth lists is the equivalent of iterating replist.  So for example, 
  
    repist(1,4);     // returns [1,1,1,1], a plain vector
    replist(replist(1,4),3);   // returns [[1,1,1,1],[1,1,1,1],[1,1,1,1]], a 3x4 structure (or is it 4x3?)
    replist(replist(replist(1,4),3),2);  // returns something I'm not going to type, 2x3x4

So my question was whether building in this type of iteration would be a useful extension, or whether the function does enough as it stands.  It actually seems like people in OpenSCAD shouldn't need rank 8 tensors and the extra generality may be unnecessary.

revarbat, 2019-03-28:
Hmm.  This seems to save exactly three characters over a list comprehension.

```
replist(1, 4)
[for(i=[1:4]) 1]
```

```
replist([1,2,3], 3)
[for(i=[1:3]) [1,2,3]]
```

revarbat, 2019-03-28:
The win becomes more reasonable for N dimensional arrays, maybe.  Though it's still a short list comprehension.

```
[for(i=[1:3]) [for(j=[1:3]) 1]]
```

revarbat, 2019-03-28:
Admittedly, the list comprehensions are a fair bit less easy to read the intent of.

revarbat, 2019-03-31:
Implementation:
```
// Function: replist()
// Usage:
//   replist(val, n)
// Description:
//   Generates a list or array of `n` copies of the given `list`.
//   If the count `n` is given as a list of counts, then this creates a
//   multi-dimensional array, filled with `val`.
// Arguments:
//   val = The value to repeat to make the list or array.
//   n = The number of copies to make of `val`.
// Example:
//   replist(1, 4);        // Returns [1,1,1,1]
//   replist(8, [2,3]);    // Returns [[8,8,8], [8,8,8]]
//   replist(0, [2,2,3]);  // Returns [[[0,0,0],[0,0,0]], [[0,0,0],[0,0,0]]]
function replist(val, n, i=0) =
    is_scalar(n)? [for(j=[1:n]) val] :
    (i>=len(n))? val :
    [for (j=[1:n[i]]) replist(val, n, i+1)];
```

revarbat, 2019-03-31:
Merged into master.

closed 2019-03-31 by revarbat