revarbat/BOSL #9: setting "MODEL_MAX_SIZE" for top_half() and other similar functions

adrianVmariano, 2019-03-24:
It seems to me sort of annoying to have to specify the size on every call for these.  Would it make sense to create some global parameter that would set this for a given model?  

If I put something like:

MODEL_MAX_SIZE = 300;

at the top of my model that could be used to adjust this (and also other types of operations where one wants to, for example, subtract the whole model from a cube).

revarbat, 2019-03-25:
I do have a precedent in `PRINTER_SLOP` in `BOSL/constants.scad`

revarbat, 2019-03-27:
I'll think about this one.  Easy to implement, but low priority.

revarbat, 2020-03-07:
I'm not going to do this in BOSL1.  In BOSL2, the defaults were increased to a large number.

closed 2020-03-07 by revarbat