revarbat/BOSL #16: Add libfile for regulat polyhedrons

revarbat, 2019-03-25:
As per suggestion from @adrianVmariano, We should have a regular solids libfile.  At the least for the platonic solids, and typical dice polyhedra: 4, 6, 8, 10, 12, 20, 30 sides. Maybe some of the Archimedean solids.

revarbat, 2019-03-25:
Polyhedra devel work by @adrianVmariano:
Requires Oskar Linde's Hull to work.  (#18)

```
use<BOSL/math.scad>
use<BOSL/transforms.scad>
       
function is_member(value, list) =
    len(list)==0 ? false :
    list[0] == value ? true : is_member(value,cdr(list));

function even_permutations(v) = [v, [v[2], v[0], v[1]], [v[1],v[2],v[0]]];

function all_permutations(v) = [v, [v[2], v[0], v[1]], [v[1],v[2],v[0]], [v[1],v[0],v[2]],[v[2],v[1],v[0]],[v[0],v[2],v[1]]];

function listcomp(a,b,n=0) =   !is_list(a) && !is_list(b) ? scalarcomp(a,b):
                               len(a)<=n && len(b)<=n ? 0 :
                               len(a)<=n ? 1 :
                               len(b)<=n ? -1 :
                               a[n] < b[n] ? -1 :
                               a[n] > b[n] ? 1 :
                               listcomp(a,b,n+1);

function scalarcomp(a,b) = a<b ? -1 : a>b ? 1 : 0;

 function quicksort(arr) =  
    !(len(arr)>0) ? [] :  
        let(  pivot   = arr[floor(len(arr)/2)],
              lesser  = [ for (y = arr) if (listcomp(y,pivot)==-1) y ],
              equal   = [ for (y = arr) if (listcomp(y,pivot)==0) y ],
              greater = [ for (y = arr) if (listcomp(y,pivot)==1) y ]
        )
        concat( quicksort(lesser), equal, quicksort(greater) );  

function unique(points) = let(sorted=quicksort(points))
    [for(i=[0:len(sorted)-1]) if (i==0 || sorted[i]!=sorted[i-1]) sorted[i]];

// sign=="even" means an even number of minus signs (odd number of plus signs)
// sign=="odd" means an odd number of minus signs (even number of plus signs)
  
function point_ref(points, sign="both") =
  unique([ for(i=[-1,1],j=[-1,1],k=[-1,1]) if (sign=="both" || sign=="even" && i*j*k>0 || sign=="odd" && i*j*k<0) 
     each [for(point=points) vmul(point,[i,j,k])]
     ]);

function cosh(x) = (exp(x)+exp(-x))/2;
function acosh(x) = ln(x+sqrt(x*x-1));

// Polyhedra data from Wikipedia and http://dmccooey.com/polyhedra/

phi = (1+sqrt(5))/2;
tribonacci=(1+4*cosh(acosh(2+3/8)/3))/3;

pname = 0;
class = 1;
facecount = 2;
edgelen = 3;
in_radius = 4;
mid_radius = 5;
out_radius = 6;
volume = 7;
vertices = 8;

library = [
  ["tetrahedron", "platonic", 4, 2*sqrt(2), sqrt(6)/12, sqrt(2)/4, sqrt(6)/4, 1/6/sqrt(2),
                   point_ref([[1,1,1]], sign="even")],
  ["cube", "platonic", 6, 2, 1/2, 1/sqrt(2), sqrt(3)/2, 1,
                   point_ref([[1,1,1]])],
  ["octahedron", "platonic", 8, sqrt(2), sqrt(6)/6, 1/2, sqrt(2)/2, sqrt(2)/3,
                   point_ref(even_permutations([1,0,0]))],
[... 219 lines cut ...]
for(i=[0:len(library)-1]) {
  place_copies([[3*i,0,0]])              // Plain polyhedron
    regular_polyhedron(index=i, mr=1);
  place_copies([[3*i,3.5,0]]){           // Inner radius means sphere touches faces of the polyhedron
    sphere(r=1.005);                     // Sphere is slightly oversized so you can see it poking out from each face
    %regular_polyhedron(index=i, ir=1);
    }
  place_copies([[3*i,7,0]]){             // Mid radius means the sphere touches the center of each edge
    sphere(r=1);
    %regular_polyhedron(index=i, mr=1);
    }
  place_copies([[3*i,11,0]]){            // outer radius means points of the polyhedron are on the sphere
    %sphere(r=.99);                      // Slightly undersized sphere means the points poke out a bit
    regular_polyhedron(index=i, or=1);
    }
}

/* regular_polyhedron(faces=60,index=0); */

// TODO
// Use volume info?
// Are archimedean & catalan solids in the "right" order in the list?
// Add a stellate function

```

adrianVmariano, 2019-03-25:
One thing I'm not sure about is whether there are other good mechanisms I should add for selecting which polyhedron you want.  Right now you can narrow by "type" and by number of faces and then index into the remaining list, or you can give a name.  I thought about allowing alternate names but it wasn't clear there were any alternate names that would be helpful.  

Catalan and Archimedean solids are duals of each other and at the very least, the ordering should be such that the nth catalan solid is the dual of the nth archimedean one on the list.  I think that might not be true at the moment.  

It's obviously pretty easy to add more convex polyhedra if you find any you think are essential---we just need to construct a vertex list and hull() does the rest.  But my list is pretty extensive, and I think it includes everything you asked for above.    

I'm not absolutely certain that all the volume data is correct, as I don't have a good way to validate it, and sometimes it was given for sizes other than the unit polyhedron and I had to adjust it.  The radii are easily checked by examining the rendered sphere and polygon together.  Is the volume data useful?  I saw a library that let you construct polyhedra with a desired volume---I could add that.  It seems not very different from using the mid-radius.

revarbat, 2019-03-25:
I have to assume the volume data is useful to *someone*, but it's not really a high priority.

adrianVmariano, 2019-03-26:
Volume data is there already.  I just don't have a way of checking its correctness, whereas I did for the radii data.  So the volume data might have errors, whereas I'm pretty sure that the radii are all correct.  (You can inspect the demo result with the array of 31 polyhedra and see by inspection that the radii are correct, or at least very close to correct by how the spheres intersect the solids.)   I suppose drastic errors in the volume data would also be pretty obvious.   The extra code to generate a polyhedron with a desired volume is just a couple lines, easy to add.  

I do think that the highest priority is providing good methods for people to actually select which shape they want.  Some of the names are so long and complex that the only way most people would get them right would be by cutting and pasting from the list in the manual.  

Basically all that happens in this code is you select a polyhedron, the code has a vertex list, it passes it to hull to get the faces.   There's a bit of scaling to control the desired size.  The core of it all is the "library" data structure which has magical radii values the number of faces, and the actual vertex list.  That list is created in most cases by permuting and reflecting a small number of generator points using functions (that should be private or "hidden").  I need unique() to avoid duplicating points, which the hull() function doesn't seem to handle.   So programmatically there's really not much going on.

revarbat, 2019-03-26:
Also requires `quicksort()` and `unique()`.  (#11)

adrianVmariano, 2019-03-26:
Technically it only requires unique() which can be written without a sort function using a loop with search().  I'm not sure if that's a good idea, but it's possible.  

You asked above for all the dice, and I realized that the 10-sided die is not included in my file.  It's one member of the infinite family of trapezohedrons.  It's not clear to me if families like this should have their own functions, or if we should have a list of special cases for such infinite families and have polyhedra handle it.   Other infinite families include the bipyramids and antiprisms.   There seem to be canonical lengths to these things that are a mystery to me.    I'm inclined to think making separate modules makes more sense.

revarbat, 2019-03-28:
Hmm.  A couple thoughts after playing around with this... 

1. While comprehensive, this is really slow.  I mean, it shouldn't take 2 to 3 seconds to generate a platonic solid.
2. Orienting these polyhedra so they are in a useful orientation may be a pain.  As designed, many polyhedra do not have a face flat to the bottom or sides, which I suspect will be a wanted orientation.
3. Yeah, finding the polyhedra will be a pain, but that can mostly be solved with documentation, with images and polyhedra names.

adrianVmariano, 2019-03-28:
1.  Not sure what is going on for you with regards to speed.  I just ran 

    for(i=[0:60])
         up(2*i) regular_polyhedron(faces=60,index=0,or=1); 

which generates 60 polyhedra that have 60 faces each and it ran in 1s on my machine.   The demo, which shows all solids with their insphere, midsphere and outsphere runs in 4 seconds and makes 124 polyhedra, of which 93 are intersected with a sphere.  I don't think you can expect faster.  

Now if you want rounded edges, so you pick r>0, and you choose a large $fn I think there's no help for it.  All the code is doing is a hull of spheres.  How would you make that faster?   I did four tetrahedra with $fn=64 and it took 27s, or 6.75s per tetrahedron.   I'm told by the experts that union is actually much slower than  hull, so it might be possible to make this faster by computing our own point sets for the spheres, unioning the points together, and then doing a hull on the resulting large point set.  Is it worth the complexity?   Linde's hull() function hits a recursion limit after a few thousand points, so I think it would be necessary to use the hack of invoking polyhedron with bogus faces and passing that to hull(). That hack is actually quite fast and computes the hull of 40000 points in 2s.   I guess union is slow because it has to check all the faces for intersection?

Actually I've been waiting for this code:
    $fn=64;polyhedron(faces=60,index=0,or=1,r=.1); 
to finish and it's taking an extremely long time---just finished at 8.5 minutes.   Maybe it *is* worth investigating the more complex alternative hack I proposed above.   Any ideas on a good way to generate a set of points on a sphere?   Do we know what points OpenSCAD uses when you ask for a sphere with a given $fn?  

2.  Yes, orientation is an issue.  One solution would be to hard code into the library a rotation vector that puts a face down.  Another solution would be to automatically find a vector normal to a face and either pass it back to the user for use with an attach library or use an attach library internally to spin the shape into position.    I wanted to do normal finding for all faces so I could do stellation anyway, so this seems like probably the better approach.   Actually it occurs to me that getting just any normal is really easy:  just compute a normal to the first face.  The stellation problem is harder because I want the face coordinates and the hull() function returns the triangulation of the faces, so it's necessary to reassemble them.  

A complication for the orientation problem is that for shapes with multiple types of face, which one do you want down?  It might be desirable to supply options for each face type.   Doing this automatically would require processing all of the faces and then looking for different ones.  

The last question is, do we change the default behavior so that shapes appear by default with a side down?  (Is there some advantage in the canonical orientation?  There might be...I don't know the answer.  It might place shapes in a meaningful relation relative to their duals or relative to each other.)  

A final question relates to raising the shapes so that they are sitting on the xy plane.   The user may not want this because the user wants to know where the shape center is, so it seems both options are needed (e.g. center=true / center=false)  But if center=false, the user may need access to the shape's distance below the xy plane for later operations.  See below on the issue of having functions that interrogate the library for the user.  

A last observation is that due to the non-sphericality of spheres, it may be very tricky to get the shape exactly lined up onto the xy plane when using r>0.   I'm not sure exactly how much of an issue this is.  Maybe everything just lines up because spheres behave reasonably in the axes directions.  

3.  I wonder if there should be a way to interrogate the library to extract information from it.  For example, you might want to ask the library how many 60 sided shapes are available, and then you could write an appropriate loop to display all of them.   Or as noted above, you want to know how far below the xy plane your shape extends---in the case of a face down orientation this will be the inner radius (for shapes that have only one type of face).

adrianVmariano, 2019-03-28:
Hmmm.  Actually running minkowski is much faster than hull:

    minkowski(){
      sphere($fn=256,r=.2);
      regular_polyhedron(faces=60,index=2,side=1);
    }

It's taking 4s to produce 60 sided solids.  Tetrahedron in 2s.  The hull hack might be faster, but is this fast enough? 

So then do I rewrite the code to use minkowski, or just take out the smoothing option entirely and let users use minkowski?   (I'm leaning towards the first option.)

adrianVmariano, 2019-03-28:
Curiously I got very different run time behavior on a different machine running linux.   The minkowski method rounds a 60 sided polyhedron in 0s.  I implemented the hack I mentioned and it takes 5s.   Not sure what to make of that.  (Hard to be sure of apples to apples comparison since the hack has a different point layout on the sphere, and maybe a different number of points.)  

In any case, we might want this other hull function since it can handle large data.   We could call this one point_hull and Linde's function we could call hull_faces, since that's really what it does.  

    module pointhull(points){
      points = simplify3d_path(points);  // colinear points are not on the hull and generate a warning message
      extra = len(points)%3;
      list = concat(
                  [[for(i=[0:extra+2])i]],
  		[for(i=[extra+3:3:len(points)-3])[i,i+1,i+2]]);
      hull() polyhedron(points, faces=list);
    }

and then here's the tweaked version of regular_polyhedron that uses it:

    function spherept(center, r, N) =
     [for(i = [0:N])
       let (theta = 180*i/N, count=1+ceil(N*sin(theta)))
         each [for(j=[0:count]) let(phi=360*j/count)
             center+r*[sin(theta)*cos(phi),sin(theta)*sin(phi), cos(theta)]]];

    module regular_polyhedron(name=undef,index=undef,type=undef,faces=undef,side=1,ir=undef, mr=undef, or=undef, r=0)
    {
      libsize = len(library);

      argcount = sum([is_def(ir)?1:0, is_def(mr)?1:0, is_def(or)?1:0]);
      assert(argcount<=1, "You must specify only one of 'ir', 'mr' and 'or'");

      indexlist = [for(i=[0:len(library)-1]) if ((is_undef(name) || library[i][pname]==name)
                                                && (is_undef(type) || library[i][class]==type)
                                                && (is_undef(faces) || library[i][facecount]==faces)) i];

      echo("indexlist = ",indexlist);
      entry = library[is_def(index) ? indexlist[index] : indexlist[0]];

      echo("index = ",library[is_def(index) ? indexlist[index] : indexlist[0]]);

      
      scalefactor = (argcount == 0 ? side :
                    is_def(ir) ? ir/entry[in_radius] :
                    is_def(mr) ? mr/entry[mid_radius] : or/entry[out_radius]) / entry[edgelen];
      if (r==0) 
          polyhedron(scalefactor*entry[vertices], hull(entry[vertices]));
      else {
        fn = segs(r);
        fixed_r = r/cos(180/fn);
        adjusted_scale = scalefactor - fixed_r / entry[in_radius] / entry[edgelen];

        pointhull([for(v=adjusted_scale*entry[vertices]) each spherept(v,fixed_r, fn)]);
           
    //      place_copies(adjusted_scale*entry[vertices])
    //        sphere(r=fixed_r, $fn=fn);

      }
    }

So which one is fastest for you?

revarbat, 2019-03-28:
> So then do I rewrite the code to use minkowski, or just take out the smoothing option entirely and let users use minkowski? (I'm leaning towards the first option.)

I concur.  However, I'd suggest naming the `r` rounding radius to something else, as `r` would be my very first instinct to use to define the outer radius of the polyhedron.  (Just like `cylinder()`, `sphere()`, and `circle()`.)

revarbat, 2019-03-28:
Actually, if we're going with consistency with the rest of the library, I'd recommend providing `r` and `d` alternate arguments for the outer radius, and a `circum` argument that would tell it that `r` or `d` are the radius to circumscribe, vs. inscribe.

Naming the rounding radius `fillet` would be consistent, but I'd be happy with `round` or `rounding`.  In BOSL2, I plan to change most references to `fillet` to `round`.

revarbat, 2019-03-28:
>  Do we know what points OpenSCAD uses when you ask for a sphere with a given $fn?

Yes.  I had to figure this out when making `staggered_sphere()` in shapes.scad.  It generates points on $fn/2 latitudes, and $fn longitudinal angles.  Something like:

```
function spherepts(r, fn) = let(tstep = 360/fn, pstep = 180 / floor(fn/2)) [
    for (phi=[pstep/2:pstep:180], theta=[0:tstep:360-tstep/2]) spherical_to_xyz(r,theta,phi)
];
```

revarbat, 2019-03-28:
Regarding finding bounds of a polyhedron, you can just pass the vertexes to pointlist_bounds()

revarbat, 2019-03-28:
Regarding spinning a face to be flat, if you have the normal for a face you want to rotate to vertical, you can just use `rot(from=normv, to=V_UP) ...`

revarbat, 2019-03-28:
I've been using `orient_and_align()` in a lot of the shapes code, with `orient` and `align` arguments, but that's mostly for cubic alignments and rotations.  The face orientation of general polyhedra is a slightly different matter, though.  Technically, orient can be any [X,Y,Z] rotation, but finding those angles is somewhat tricky.

revarbat, 2019-03-28:
Regarding slowness, the 2 to 3 second generation time was for a plain platonic dodecahedron (12 pentagonal faces) with no rounding at all.  Mind you, my laptop is not very fast.

revarbat, 2019-03-28:
Looking at the polyhedra, a lot of shapes don't actually have an obvious face to orient flat. I'm not sure what to do about this.

revarbat, 2019-03-28:
> for(i=[0:60]) up(2*i) regular_polyhedron(faces=60,index=0,or=1);

This took 4 seconds for me.

adrianVmariano, 2019-03-28:
So it takes 2-3 s to make an icosahedron and 4 s to make 60 of the 60-face polyhedron?  That's weird.   

In order to use rot(from, to..) I need to have made geometry.  But if I then want to use pointlist_bounds I need to still have coordinate points.  You don't seem to have the from and to options on the operators that work on points.   But also, for the case of no rounding, the in-radius should give the correct distance.  But with rounding, it might be slightly in error---that's where I would end up not exactly on the x-y plane, I think.  I suppose if I do rounding using the point_hull method I'd have the point set.    

Any thoughts on rounding via minkowski vs rounding via point_hull?  

I'm not sure I understand your suggestion for inputs r and d.  First I thought just make r equivalent to or.  But you were thinking another argument, say rtype="outer", "middle","inner" ?  

I don't think orient_and_align() will help with polyhedra not based on the right angle.  I think the main challenge really is to decide exactly what behavior makes sense, and then I can find some way to implement the desired behavior, most likely by constructing the faces and then picking a suitable face to supply an alignment vector.   One possibility is that in face-down mode, the largest face goes down.  But maybe it's desirable to let the user pick, so then we need an input mechanism to specify which face goes down, perhaps by number of sides?  facedown=5 puts the pentagon down and facedown=3 puts the triangle down?   As you say, there are also some shapes where it really isn't clear that face down makes a lot of sense...but you can also argue that corner down often doesn't make sense, so I'm not sure this should be a concern.  The user can choose the facedown option or not.  

A question about centering is whether the uncentered shape should be entirely in the positive octant like happens e.g. with cube().

adrianVmariano, 2019-03-29:
Another question: do we need to supply the user with face coordinates and normals...or with some other mechanism for putting stuff onto faces or polyhedra.  (Maybe regular_polyhedron does something with its children?)

revarbat, 2019-03-31:
Linking in #23 as a probable requirement.

adrianVmariano, 2019-03-31:
I thought about this more and I think I'm going to try to do something like what you did with grid, where the children get placed on each face of the polyhedron, and I'll put the face corners into a $variable for the user to use if desired.   The one thing I'm not sure about is how to order the faces.

Another question is whether it's better to supply a "difference=true" operation that makes it subtract the children from the polyhedron or supply a "draw=false" option that causes it to place the children without drawing the polyhedron (so the user could create the solid and then difference the children).  Any thoughts here?

revarbat, 2019-04-01:
I kind of like the $variable face children idea.  Each one will probably need to know number of face sides and size, face normal, and more.

Regarding `difference` and `draw`, they both seem to have their uses.  I see three usages:

1. Tessellating a polyhedron.
2. The inverse, like in the great dodecahedron.
3. As a way to place children, without the polyhedron itself.

Feels like it should be a single argument, with three values.

revarbat, 2019-04-01:
Yes, orient_and_align() seems inappropriate here.

I think I prefer `minkowski()` for rounding.  It'll probably be faster than hulling the point cloud.

`rotate_points3d()` now accepts `from`, `to`, and `axis` args.

In `cyl()`, you can specify radius `r` or diameter `d`, then the `circum` argument specifies whether you want in inscribe or circumscribe the circle of that radius/diameter.  For purposes of polyhedra, I can see it having true, false, and "mid" optional values.

I really don't know what to do about polyhedra orientation.  Some shapes make sense to have large face down.  Some make sense to have point-up, with no particular face aligned.

Centering is also a problem.  The center of tetrahedral shapes, for example is not vertically centered.

adrianVmariano, 2019-04-05:
I've implemented a bunch of new stuff.  Hulling the point cloud with the polyhedron hack is really fast.  I can hull 40000 points in a few seconds.  But it seems to be system dependent which one is faster.  One one of my computers it seems like they were about the same.  The other one, the hull method was twice as fast.  I would guess that the time to compute the points of all the spheres is a substantial fraction of the time.   I was feeling like having to compute the points on the sphere makes the point hull method a little bit messier.   It does have the advantage that we can get the resulting shape exactly onto the xy plane, which is otherwise impossible.  

I've added "r" and "d".  I don't like "circum" for this context because it doesn't make sense really to write circum="middle".  I made an "rtype" parameter that can be set to "outside", "inside" or "middle".  I am not convinced this is worth keeping because of or, ir and mr.   (Note also you might want to clarify the purpose of circum for cyl() in the wiki.  I had to think for a bit before I understood what you meant by "circumscribe a circle".   I would add a note about how circles are polygons and their interiors are undersized.)  

I added align which works like for cubes---it shifts the polyhedron so that it's most extreme point is on the specified plane.  

I added facedown which puts some random face down.  This needs help as sometimes it's really not the right face.   More on that later.  If you use facedown you can shift the solid up so it's resting on the xy plane.  

The code now puts the children onto the faces and it passes $face and $faceindex to the child.   I added a "draw=false" option so you can place objects without the polyhedron (or for differencing from a drawn solid).   If you have repeat=true then the children are cycled through to cover all faces.   The faces are in an arbitrary order.  

The obvious problems I see that need help are

* What should face order be for children?
* full_faces is the function that assembles faces from triangles and it's really slow.  I put it into the main call and the time for the 124 shape demo went from 3 seconds to 10 minutes.   One possible optimization would be not to call it when the faces are all triangles.  But that will only help some.  If you have any thoughts on how to optimize it let me know.  Basically it collects coplanar triangles together.  
* I had the idea that you could say facedown=5 to request a pentagon down, or facedown=8 to request octagon down.  This would be easiest of I could make full_faces run faster.    
* rot(from,to) gives a random orientation of the face.  I would like to control this so an edge is parallel to the x axis.   Take a look at the tetrahedron with numbers.  This requires a better function for selecting the rotation---something more like obiscad's attach operator.  
* I was just thinking of adding to the library the type of faces in each polyhedron, which would enable another selection criterion (show solids that have pentagons) and would enable the optimization of avoiding full_faces in the case of all triangular faces.  

```
use<BOSL/math.scad>
use<BOSL/paths.scad>
use<BOSL/transforms.scad>
use<hull.scad>

function num_defined(v) = sum([for(entry=v) is_def(entry) ? 1 : 0]);

function sortind(data) = let(augdata = array_zip([for(i=[0:len(data)])i], data))
                                                    array_subindex(sort(augdata,1),0);

function even_permutations(v) = [v, [v[2], v[0], v[1]], [v[1],v[2],v[0]]];
function all_permutations(v) = [v, [v[2], v[0], v[1]], [v[1],v[2],v[0]], [v[1],v[0],v[2]],[v[2],v[1],v[0]],[v[0],v[2],v[1]]];

// sign=="even" means an even number of minus signs (odd number of plus signs)
// sign=="odd" means an odd number of minus signs (even number of plus signs)
  
function point_ref(points, sign="both") =
  unique([ for(i=[-1,1],j=[-1,1],k=[-1,1]) if (sign=="both" || sign=="even" && i*j*k>0 || sign=="odd" && i*j*k<0) 
     each [for(point=points) vmul(point,[i,j,k])]
     ]);

// Polyhedra data from Wikipedia and http://dmccooey.com/polyhedra/

phi = (1+sqrt(5))/2;
tribonacci=(1+4*cosh(acosh(2+3/8)/3))/3;

pname = 0;
class = 1;
facecount = 2;
edgelen = 3;
in_radius = 4;
mid_radius = 5;
out_radius = 6;
volume = 7;
vertices = 8;

library = [
  ["tetrahedron", "platonic", 4, 2*sqrt(2), sqrt(6)/12, sqrt(2)/4, sqrt(6)/4, 1/6/sqrt(2),
                   point_ref([[1,1,1]], sign="even")],
  ["cube", "platonic", 6, 2, 1/2, 1/sqrt(2), sqrt(3)/2, 1,
                   point_ref([[1,1,1]])],
  ["octahedron", "platonic", 8, sqrt(2), sqrt(6)/6, 1/2, sqrt(2)/2, sqrt(2)/3,
                   point_ref(even_permutations([1,0,0]))],
  ["dodecahedron", "platonic", 12, 2/phi, sqrt(5/2+11*sqrt(5)/10)/2, (3+sqrt(5))/4, sqrt(3)*phi/2, (15+7*sqrt(5))/4,
                   point_ref(concat([[1,1,1]],even_permutations([0,phi,1/phi])))],   
  ["icosahedron", "platonic", 20, 2, phi*phi/2/sqrt(3), cos(36), sin(72), 5*(3+sqrt(5))/12, 
                   point_ref(even_permutations([0,1,phi]))],
  ["truncated tetrahedron", "archimedean", 8, sqrt(8), sqrt(6)/4, 3*sqrt(2)/4, sqrt(11/8), 23*sqrt(2)/12, 
                   point_ref(all_permutations([1,1,3]),sign="even")],         
  ["cuboctahedron", "archimedean", 14, sqrt(2), sqrt(2)/2, sqrt(3)/2, 1, 5*sqrt(2)/3, 
                   point_ref(all_permutations([1,1,0]))],
[... 336 lines cut ...]
// The canonical coordinates may have useful relationships, 
[... cut ...]
    points = simplify3d_path(points);  // colinear points are not on the hull and generate a warning message
    extra = len(points)%3;
    list = concat(
                [[for(i=[0:extra+2])i]],
		[for(i=[extra+3:3:len(points)-3])[i,i+1,i+2]]);
    hull() polyhedron(points, faces=list);
}

/// tetrahedron speed test
/*
place_copies([for(i=[0:20])[0,0,i]])
regular_polyhedron("tetrahedron", align=[1,1,1]);
*/

// TODO
// Use volume info?
// Are archimedean & catalan solids in the "right" order in the list?
// Add a stellate function

adrianVmariano, 2019-04-05:
You think it's worth adding a difference flag where the children are subtracted from the solid?  Is there a way to do this without having to repeat the whole children loop? 

Also I was thinking I should provide a way to have the children oriented with respect to the parent axis instead of the polyhedron surface normal.

adrianVmariano, 2019-04-05:
Oh, and also, if you're going to make a ton of polyhedra the code is going to be slow with "use".  You need to use "include" instead because recomputing library every call wastes time.   I can't believe that bug was discovered in 2014, was "easy to fix" and not just fixed then and there.

adrianVmariano, 2019-04-05:
Are you planning to add something like sortind()?  I needed this for putting the face vertices into order and I thought you were going to add it, but it looks like it's not there (yet).

revarbat, 2019-04-05:
These should do.  I'll add them.
```
fuction enumerate(v) = [for (i=[0:len(v)-1]) [i,v[i]]];
function sortind(v) = array_subindex(sort(enumerate(v),idx=1), 0);
```

revarbat, 2019-04-05:
Added `enumerate()` and `sortidx()` in master branch commit 670c47ca869a2ce520a84810b9951dc0bfbc99af

revarbat, 2019-04-05:
Huh.  I didn't even realize I started with sortind(), and ended up with sortidx().

adrianVmariano, 2019-04-10:
I managed to redesign the slow function full_faces so it is fast.   The extreme slowdown I observed was due to examples in hull.scad.   I have fixed some errors in the polyhedra names and ordered them in a well-defined and consistent order.  I added face types and allow you to select polyhedra by type of face (e.g. give polyhedra that include a pentagon or give polyhedra that have a square or octagon).   I added a "draw=false" option to suppress display of the solid itself.  I added rotate_children=false to place the children oriented in the parent coordinate system.  

Here are what I see as the remaining questions:

1. Is there a way to add a "difference" flag that doesn't require rewriting the entire loop that creates the children?  (Is it worth doing anyway?)
2. Is there a natural face order for the children
3. A way to choose which face is down when there are multiple faces of the same type (e.g. triangles that are not symmetrical with respect to each other)
4. Should rtype be retained?
5. Is it worth adding a stellate option?  It's easy enough to erect the pyramids (as shown in the examples) but I'm not sure of a good way to handle the children, then.   I could replicate the child associated with each face and put it on every face of the pyramid, I guess.  Or I suppose I could do indexing where if I erect a 5 sided pyramid on a dodecahedron that I take children(i), children(i+12), children(i+24), children(i+36) and children(i+42) and put those on the pyramid corresponding to face(i).  
6. Should I add other families of polyhedra that would be special case handled, such as the trapezohedra?  (That makes the 10 sided die)   I thought about doing a separate function for these, but there's going to be a lot of duplication of code if I do that.   Is it too much complexity to allow name="trapezohedron" and then require faces=N, side=L and maybe a length or a second side spec?   
7. Should there me a method for getting the Kepler-Poinsot Solids (which as you can see in the examples I've produced all four of)---but they could have special case polyhedron names as well.  

```
use<lib/BOSL/math.scad>
use<lib/BOSL/paths.scad>
use<lib/BOSL/transforms.scad>
use<hull.scad>

function num_defined(v) = sum([for(entry=v) is_def(entry) ? 1 : 0]);

function quantvect(vec,scale) = [for (v=vec) quant(v,scale)];

/////////////////////////////////////////////////////////////////////////////
//
// Some internal functions
//
// All permutations and even permutations of three items
//
function _even_perms(v) = [v, [v[2], v[0], v[1]], [v[1],v[2],v[0]]];
function _all_perms(v) = [v, [v[2], v[0], v[1]], [v[1],v[2],v[0]], [v[1],v[0],v[2]],[v[2],v[1],v[0]],[v[0],v[2],v[1]]];
//
// Point reflections across all planes.  In the unconstrained case, this means one point becomes 8 points. 
// 
// sign=="even" means an even number of minus signs (odd number of plus signs)
// sign=="odd" means an odd number of minus signs (even number of plus signs)
//
function _point_ref(points, sign="both") =
  unique([ for(i=[-1,1],j=[-1,1],k=[-1,1]) if (sign=="both" || sign=="even" && i*j*k>0 || sign=="odd" && i*j*k<0) 
     each [for(point=points) vmul(point,[i,j,k])]
     ]);

// Polyhedra data from Wikipedia and http://dmccooey.com/polyhedra/

phi = (1+sqrt(5))/2;
tribonacci=(1+4*cosh(acosh(2+3/8)/3))/3;

pname = 0;
class = 1;
facecount = 2;
facevertices = 3;
edgelen = 4;
in_radius = 5;
mid_radius = 6;
out_radius = 7;
volume = 8;
vertices = 9;

library = [
  // Platonic Solids

  ["tetrahedron", "platonic", 4,[3], 2*sqrt(2), sqrt(6)/12, sqrt(2)/4, sqrt(6)/4, 1/6/sqrt(2),
                   _point_ref([[1,1,1]], sign="even")],
  ["cube", "platonic", 6, [4], 2, 1/2, 1/sqrt(2), sqrt(3)/2, 1,
                   _point_ref([[1,1,1]])],
  ["octahedron", "platonic", 8, [3], sqrt(2), sqrt(6)/6, 1/2, sqrt(2)/2, sqrt(2)/3,
                   _point_ref(_even_perms([1,0,0]))],
  ["dodecahedron", "platonic", 12, [5], 2/phi, sqrt(5/2+11*sqrt(5)/10)/2, (3+sqrt(5))/4, sqrt(3)*phi/2, (15+7*sqrt(5))/4,
                   _point_ref(concat([[1,1,1]],_even_perms([0,phi,1/phi])))],   
  ["icosahedron", "platonic", 20, [3], 2, phi*phi/2/sqrt(3), cos(36), sin(72), 5*(3+sqrt(5))/12, 
                   _point_ref(_even_perms([0,1,phi]))],
[... 481 lines cut ...]
    list = concat(
                [[for(i=[0:extra+2])i]],
		[for(i=[extra+3:3:len(points)-3])[i,i+1,i+2]]);
    hull() polyhedron(points, faces=list);
}

/// tetrahedron speed test

//place_copies([for(i=[0:20])[0,0,i]])
//regular_polyhedron(index=30, align=[1,1,1]);

//for(i=[0:40]) up(i*10)
//               regular_polyhedron(index=30, side=i/2);

// TODO
//
// Use volume info?
// Add a stellate function
// Support choosing a face number down
// Support multiple inspheres/outspheres when appropriate?
// trapezohedron?
//
// face order for children?
// orient faces so an edge is parallel to the x-axis
```

revarbat, 2019-04-10:
I fear I may end up complicating your life some.

What I’m working on now uses `orient_and_align()` to allow `attach()`ing objects together, similarly to the Relativity library.  If you pass a list of connectors [“name” XYZPOS DIRVEC ROTANG] in via the `alignments` argument, you can make it possible to attach objects to a face by name. (Is: “face3”)

I’m still debating whether to check my code in now, of if I should start making the BOSL2 cleanup including these.

revarbat, 2019-04-10:
Since we pretty much have to use `include` instead of `use`, can you put a prefix before the constants you declare?

revarbat, 2019-04-10:
I think at this point we should focus on providing a usable polyhedron library file rather than an absolutely complete one. I don’t really have an answer for many of your questions. I’m kinda trusting you to figure out something usable with this. I’ll focus on getting Linde’s Hull added.

adrianVmariano, 2019-04-10:
I'll prefix my constants with "_" the same as internal functions.  Is that good enough?  

I suggest naming Linde's hull function as hull_faces() since that's really what it does.   Should we combine the hull_faces function and my stuff into one polyhedra.scad file?   I also think it is worth supplying the hull_points module that uses polyhedron with bogus faces.  

I believe that my polyhedra code is in a usable state at present.  I suppose the main question would be whether to keep "rtype", since that's an interface issue and we wouldn't want to delete it later after release.   I'm not sure about the implications of your new attachment stuff.

revarbat, 2019-04-10:
A _ prefix is probably fine, though constants should be all caps for consistency. I can see having hull_faces() in the polyhedra libfile, if you want to take on cleaning it up for using BOSL functions. I have no strong feelings about rtype. Feel free to axe it.

revarbat, 2019-04-10:
The new attachment code lets you do things like:
```
cube(100, align=“bottom”) {
    attach(“left”, “bottom”) cylinder(d=40, h=100);
    attach(“top-front-right”) sphere(d=50);
}
```

You’re already using children with polyhedra, though, so you may not be able to integrate with attach().

adrianVmariano, 2019-04-10:
So does "top-front-right" mean the sphere is centered on the corner of the cube?   How do you specify rotation of the attached object (for situations where it lacks rotational symmetry)?    

The situation where I felt like attachment would really help was when I had made a sub-part and had a vector for maybe the back corner and an up direction, and then I want to stick it on my complex model at an edge, or maybe 2 units in from the edge oriented correctly.  Figuring out the right sequence of rotations to do this always seems annoying.   It could also be useful for differencing roundover masks.   It might be the case that just extending the rot() from and to operations so that they also respect a rotation solves this problem (though I'm not sure about the translation part).  

It's a little unclear what sort of  "attach" commands would make sense for the polyhedra.   I could do something like attach(3) to put something on face 3.  I'm not sure it's worth the effort to do something like this.  

Do you think any of the functions that are in hull() should be added to math.scad for general use?  Those functions are area2d (area of a triangle), colinear (2d, by checking triangle area), and distance to a plane, coplanar, and then max_index and min_index (which return the index of the extreme value in a list).  

I realized that the hull() function will also do hull in 2d and 1d, which makes me question hull_faces() as a good name for it.

revarbat, 2019-04-10:
`attach()` and `align` can be one of the standard connectors, such as "center", "left", "back-top", or "right-front-bottom".  The rotations for each one are consistent with the ORIENT_\* orientations, so up is generally towards Z+, except on the top, center, and bottom, where up is towards Y+.  For non-standard connectors, the rotation is specified in the connector info passed to orient_and_align(), along with attachment point and directional vector.

If you don't want that orientation rotation on attachment, you can attach with `norot=true` and your attached part will be vertically oriented in its original rotation.  If you want the vector of the attachment, but not the rotation, you can tweak it with `zrot()`, since you'll be starting with a known rotation.

Here's a prismoid with the standard connectors shown:
[image]

Just for fun, here's the code to generate a fractal tree.
```
include <BOSL/constants.scad>
include <BOSL/transforms.scad>
include <BOSL/primitives.scad>
include <BOSL/beziers.scad>                                            
                                                                       
module leaf(s) {                                                       
    path = [                                                           
        [0,0], [1.5,-1],                                               
        [2,1], [0,3], [-2,1],                                          
        [-1.5,-1], [0,0]                                               
    ];                                                                 
    xrot(90)                                                           
    linear_extrude_bezier(                                             
        scale_points(path, [s,s]/2),                                   
        height=0.02                                                    
    );                                                                 
}                                                                      
                                                                       
module branches(minsize){                                              
    if($parent_size2.x>minsize) {                                      
        attach("top")                                                  
        zrot(gaussian_rand(90,10))                                     
        zring(n=floor(log_rand(2,5,4)))                                
        zrot(gaussian_rand(0,5))                                       
        yrot(gaussian_rand(30,5))                                      
        let(                                                           
            sc = gaussian_rand(0.7,0.05),                              
            s1 = $parent_size.z*sc,                                    
            s2 = $parent_size2.x                                       
        )                                                              
        cylinder(d1=s2, d2=s2*sc, l=s1)                                
        branches(minsize);                                             
    } else {                                                           
        recolor("springgreen")                                         
        attach("top") zrot(90)                                         
        leaf(gaussian_rand(100,5));                                    
    }                                                                  
}                                                                      
recolor("lightgray") cylinder(d1=300, d2=250, l=1500) branches(5);
```

[image]

revarbat, 2019-04-10:
Oh, and yes, "top-front-left" means the upper left-front corner.  "bottom-back" means the midpoint of the rear bottom edge.  "right" means the center of the right face.

revarbat, 2019-04-10:
`gaussian_rand()` and `log_rand()` are two new math.scad functions to generate single random numbers with gaussian/normal and logarithmic distributions, respectively.

revarbat, 2019-04-10:
More fun --  differencing and intersection can be handled with tags:

```
diff("body pole", "hole")
staggered_sphere(d=100, $tags="body") {
	tags("hole") {
		cyl(d=55, h=101, realign=true, orient=ORIENT_X);
		cyl(d=55, h=101, realign=true, orient=ORIENT_Y);
	}
	cyl(d=55, h=100, realign=true, $tags="pole", orient=ORIENT_Z);
	cyl(d=15, h=140, realign=true, $tags="axle", orient=ORIENT_Z);
}
```
[image]

revarbat, 2019-04-10:
Yeah, there's several functions in Linde's hull that would be good to add to math.scad.

adrianVmariano, 2019-04-11:
Does it make sense for you to integrate the functions you want to add and then once you're done, I'll adjust Linde's function to call the library functions, sanitize the private functions and merge it into polyhedra.scad?  

Do you think that I should name the top level variables in polyhedra.scad with the expectation that users might actually refer to them?  Or treat them as private variables that are unfortunately exposed due to the limitations of OpenSCAD?   If they are private, I don't need to follow any convention other than prefixing with '_' or otherwise obscuring them.  If they are public it's a different matter.  I was thinking of writing a regular_polyhedron_info function that would pull data out of the library data structure about polyhedra so that there's no need to (intentionally) expose the library data structure to users.  

Does the documentation on the wiki get autogenerated from the comments above each function?  I'm also wondering about writing up the documentation for regular_polyhedron, which is going to be involved, with ~35 examples.

revarbat, 2019-04-11:
I can take a quick run at integrating the library-worthy functions first, sure.

If you make one or more info functions, then you don't need to expose the constants.  I do worry that some of the names seem very likely to get collisions as they are, though.  Especially `phi`, `library`, `class`, `edgelen`, `volumes`, `vertices`, and `facecount`.  Your code doesn't look like it actually needs to have most of those defined outside of the `regular_polyhedra()` module, currently.

Only `phi` and `library` seem to need to be outside `regular_polyhedra()`.  The constant `phi` looks like a standard constant, so should probably be moved to constants.scad and renamed `PHI`.  The `library` array should be renamed something like `POLYHEDRA_LIBRARY`, and if there's more temporary values than `phi` needed to generate it, you can generate the array in a function, and assign it to the constant from there.

revarbat, 2019-04-12:
Yes, the wiki docs are generated from the comments above the modules/functions by the scripts/docs_gen.py script.  Not all comments are added to the wiki.  Just those comment blocks starting with certain keywords:
`// LibFile: NAME`, or `// Section: NAME`, or `// Module: NAME` or `// Function: NAME`, or `// Constant: NAME`.

Module and Function docs blocks have a specific format.  Indentation is important, as it denoted the end of sub-block:
```
// Module: foo()
// Status: DEPRECATED, use BLAH instead.
// Usage: Usage Type
//   foo(foo, bar, [qux]);
//   foo(bar, baz, [qux]);
// Usage: Usage Type 2
//   foo(foo, flee, flie, [qux])
// Description: Short description.
// Description:
//   A longer, multi-line description.
//   All description blocks are added together.
//   You can use markdown notation as well.
//   You can end multi-line blocks by un-indenting the
//   next line, or by using a blank comment line like this:
//
// Arguments:
//   foo = This is the description of the foo argument.  All on one line.
//   bar = This is the description of the bar argument.  All on one line.
//   baz = This is the description of the baz argument.  All on one line.
//   qux = This is the description of the qux argument.  All on one line.
//   flee = This is the description of the flee argument.  All on one line.
//   flie = This is the description of the flie argument.  All on one line.
// Side Effects:
//   `$floo` gets set to the floo value.
// Examples: Each line below gets its own example block and image.
//   foo(foo="a", bar="b");
//   foo(foo="b", baz="c");
// Example: Multi-line example.
//   lst = [
//       "multi-line examples",
//       "are shown in one block",
//       "with a single image.",
//   ];
//   foo(lst, 23, "blah");
// Example(2D): Example to show as 2D top-down rendering.
//   foo(foo="b", baz="c", qux=true);
// Example(Spin): Example using rotating animation to show all faces.
//   foo(foo="b", baz="c", qux="full");
// Example(FlatSpin): Example using rotating animation to view top, front, back, left, and right faces.
//   foo(foo="b", baz="c", qux="full2");
```

Functions and constants won't generate images for examples unless you force it with a `(2D)`, or `(3D)` tag on the Example marker.  The full set of example tags are `2D`, `3D`, `Spin`, `FlatSpin`, `FR`, `Small`, `Med`, `Big`.  The `FR` tag is to force full rendering from OpenScad, if the preview mode is insufficient.  The size tags can generate larger images, if you need to show more detail.  You can combine tags like this: `// Example(FlatSpinBig):`

adrianVmariano, 2019-04-12:
I definitely wasn't thinking it is ok to expose names like "vertices" and so on.  I was assuming that we could use "use" and they would all be hidden.  But since we need to allow "include" then I can just define them inside the function.  If we are going to document the "library" data structure for user use I think I would call it POLYHEDRA.  If we're going to have it "private" I was thinking I might name it _polyhedra_lib or something like that.   I am inclined to keep it "private" and not document it, but supply access through a function (which I think will be better for a user.)  I agree that PHI is a reasonable constant to define globally, though I don't know that I want to include the constants file within my file since it pulls in a bunch of stuff the user didn't ask for.    I can call them _phi and _tribonacci.   Those are the only two parameters I use.  I could define the polyhedra library in a function.  You think that's better?   It trades off the clutter of _phi and _tribonnacci for the clutter of a function name, and the function name would clutter the name space even if "use" is used.  

I'll try to get my documentation into compliance.  Can I have multiple paragraphs within a section?  Are the block names like "Side effects" arbitrary, or they have to match a list of allowed headings?  So I can write something like

```
Examples(3D):
  regular_polyhedron("tetrahedron");
  regular_polyhedron("cube");
etc

and run through all of them and it will make an image for each one?

revarbat, 2019-04-12:
The (3D) is the default for Example/Examples for a module, so you don’t need it.  Yes, each line of an Examples block gets its own example image and code block in the docs.

revarbat, 2019-04-12:
The block names are not arbitrary. `Side Effects:` is one of the optional standard block names.

revarbat, 2019-04-12:
Really, only the Module/Function and Description are required.

revarbat, 2019-04-12:
If you have a line as just a comment with three trailing spaces `//   `, I believe it will render as a paragraph break.

adrianVmariano, 2019-04-12:
I was just thinking about how to restructure my current documentation, where I have different sections for how to select a solid, choosing size, choosing orientation, and so on, with embedded (textual) examples.   I was thinking about having all of this stuff in the Description section vs having separate sections "Selecting a Polyhedron", "Specfiying Polyhedron Size" and so on.

revarbat, 2019-04-12:
Remember, you can use markdown syntax in the description to have subsections. I’ll look into arbitrary blocks though.

revarbat, 2019-04-12:
More complete docs formatting explanation at:
https://github.com/revarbat/BOSL/blob/master/WRITING_DOCS.md

revarbat, 2019-04-12:
I've added `PHI` to math.scad.

revarbat, 2019-04-12:
Finally ran the latest polyhedra demo you sent.  Verrrrry nice.  Good job!

revarbat, 2019-04-12:
BTW, the `Display of all solids with insphere, midsphere and circumsphere` demo now times at 27 seconds on my weak-sauce laptop.

adrianVmariano, 2019-04-13:
The run time was slower than I expected, as I thought I had it at 6s.  I started fiddling and got it down to 6s on my machine (from 13s originally) but some of the optimizations seemed odd.  So I tried changing 

    use<BOSL/math.scad>

to

    include<BOSL/math.scad>

and my run time went to 4s.  Now I'm wondering if the optimizations were all just minimizing the number of library function calls.  What does your run time do if you make that change?  

Now I wonder if use has some other secret overhead.

[3 more replies]

closed 2019-05-05 by revarbat