revarbat/BOSL #18: Add Oskar Linde's hull()

revarbat, 2019-03-25:
https://github.com/openscad/scad-utils/blob/master/hull.scad

adrianVmariano, 2019-03-25:
I suggest adding hull() as part of polyhedra.scad.  We might want a secondary file to keep the namespace clean, as hull() depends on a ton of stuff we probably don't want exposed.   I also think there should be some convention for functions that we're too lazy to actually hide but which aren't intended for users to call, maybe prefixing with "_" or something like that.

revarbat, 2019-03-25:
Agreed on the `_` prefix for non-exposed functions.

adrianVmariano, 2019-03-26:
Perhaps we should rename it hull_faces() since it actually returns a face list.

There is discussion on the forum of doing a hull of points by calling the hull module on a bogus polyhedron with a vertex list and a list of garbage faces that includes all the vertices.   Apparently this outperforms Linde's code, though of course you don't get a face list to work with.

revarbat, 2019-04-12:
Working on integrating the common functions into the math library.

revarbat, 2019-04-16:
Added in master branch commit 96739c3ea038703e582b3170ceca4ad5dead7173

closed 2019-04-16 by revarbat