revarbat/BOSL #19: Add roundcorners

revarbat, 2019-03-25:
I haven't had the time to understand this yet, but as I understand the discussion around it, this is a way to provide corners with continuous rate of curvature change, making for prettier corners.

Code by @adrianVmariano

```
use<lib/BOSL/math.scad>
use<lib/BOSL/beziers.scad>

// TODO:
//   remove colinear points?
//
// roundcorners(path, curve, type, all, closed)
//
// path: a list of 2d or 3d points, possibly with an extra coordinate giving smoothing parameters, e.g.
//      ex:  [[0,0],[0,1],[1,1],[0,1]]    2d point list
//           [[0,0,0], [0,1,1], [1,1,2], [0,1,3]]    3d point list
//           [[0,0,.2],[0,1,.1],[1,1,0],[0,1,.3]]    2d point list with smoothing parameters
//           [[0,0,0,.2], [0,1,1,.1], [1,1,2,0], [0,1,3,.3]]    3d point list with smooth parameters
//           [[0,0,[.3,.5], [4,0,[.2,.6]], [4,4,0], [0,4,[
// curve: set to "smooth" to get continuous curvature 4th order bezier curves
//        set to "circle" to get circular arcs
// type: set to "cut" with either curve type to specify how much of the corner to "cut" off.  The 
//                    smoothing parameter will be the distance from the corner to the tip of the rounded curve
//       set to "radius" to use with curve=="circle" to specify the curve radius
//       set to "joint" to use with curve=="smooth" to specify the distance from the corner at which
//                    the curve joints the shape
// all: set this to a curvature parameter to apply to all points on the list.  If this is set then all
//      of the values given in the path are assumed to be geometrical coordinates.  If you don't set it
//      then the last value of each entry in path is assumed to be the smoothing parameters
// closed: set to true (the default) if the curve is closed and false if the curve is open at the ends
//
// If you select curve=="smooth" then there are two smoothing parameters.  The first one
// is the cut or joint distance as given type "type".  The second one is a curvature
// parameter which is a number in [0,1], where larger numbers have a more abrupt
// transition and smaller ones a more gradual transition.  If the curvature parameter is
// close to zero the transition is so gradual that it may take a very large distance.
//
// If you select curves that are too large to fit the code will fail with an error.  It displays a set
// of scale factors that you can apply to the (first) smoothing parameter which will reduce the size of the 
// curves so that they will fit on the shape.  If the scale factors are larger than one then they 
// indicate how much you can increase the curve size before collisions will occur.
//
// https://hackernoon.com/apples-icons-have-that-shape-for-a-very-good-reason-720d4e7c8a14

function roundcorners(path, curve, type, all=undef,  closed=true) =
 let(
   default_curvature = 0.7,   // default curvature for "smooth" curves
   typeok = type == "cut" || (curve=="circle" && type=="radius") ||
                             (curve=="smooth" && type=="joint"),
   pathdim = array_dim(path,1),
   have_all = all==undef ? 1 : 0,
   pathsize_ok = is_num(pathdim) && pathdim >= 2+have_all && pathdim <= 3+have_all
 )
 assert(curve=="smooth" || curve=="circle", "Unknown 'curve' setting in roundcorners")
 assert(typeok, curve=="circle" ? "In roundcorners curve==\"circle\" requires 'type' of 'radius' or 'cut'":
                                  "In roundcorners curve==\"smooth\" requires 'type' of 'joint' or 'cut'")
 assert(pathdim!=undef, "Input 'path' has entries with inconsistent length")
 assert(pathsize_ok, str("Input 'path' must have entries with length ",2+have_all," or ",
                         3+have_all, all==undef ? " when 'all' is not specified" : "when 'all' is specified"))
 let(
   pathfixed= all == undef ? path : array_zip([path, replist([all],len(path))]),
   dim = len(pathfixed[0])-1,
   points = array_subindex(pathfixed, [0:dim-1]),
   parm = array_subindex(pathfixed, dim),
   // dk will be a list of parameters, for the "smooth" type the distance and curvature parameter pair,
   // and for the circle type, distance and radius.  
   dk = [for(i=[0:len(points)-1]) let(  
     angle = pathangle(wrap_range(points,i-1,i+1))/2,
     parm0 = is_list(parm[i]) ? parm[i][0] : parm[i],
     k = curve=="circle" && type=="radius" ? parm0 :
[... 107 lines cut ...]
// 
function _array_dim_recurse(v) =
    // If first entry is not a list then v is either a list of singletons, in which case we return []
    // because there is no 2nd dimension, or if some entry is a list we return undef
   !is_list(v[0]) ?   
       (sum( [for(entry=v) is_list(entry) ? 1 : 0]) == 0 ? [] : [undef] ) : 
    // First entry is a list so check for length consistency of the list entries, and then
    // recursively call to get the rest of the dimensions
    let( firstlen = len(v[0]),
           first =  sum( [for(entry = v) len(entry) == firstlen  ? 0 : 1]   ) == 0 ? firstlen : undef,
           leveldown = flatten(v) )
    is_list(leveldown[0]) ? concat([first],_array_dim_recurse(leveldown)) : [first];

// 
function array_dim(v,depth) =
   // Add the first dimension entry (the length) to the rest of the values
 depth == undef ? concat([len(v)], _array_dim_recurse(v)) :
   // A depth was requested, so compute the dimension list and then return
   // the requested entry.  
 depth == 0 ? len(v) :
   let(
      dimlist = _array_dim_recurse(v)
   )
   depth > len(dimlist) ? 0 : dimlist[depth-1];
```

adrianVmariano, 2019-03-25:
Here's the summary:  It takes a path or a polygon path and rounds the corners either with circles or with continuous curvature order 4 beziers.  You can specify the roundover by how much is removed, which is much more predictable with acute corners vs specifying a radius, which can chop off an unexpectedly large amount.   The path can be 2d or 3d, and the radius can be different (or even zero) at every corner.  

The continuous curvature roundovers feel better on printed models in my testing, and may also look better, especially if you have room for a very gradual roundover.  

One major lingering issue is specifying the number of points used for the roundover.  Should it depend somehow on $fn, $fs and $fa?   It can't just be a constant because one corner might have a huge roundover that needs 500 points and another one might be a tiny roundover that just needs a dozen.

revarbat, 2019-03-26:
Perhaps it needs something like the splinesteps arguments in `beziers.scad`?

adrianVmariano, 2019-03-26:
That would mean either that the user has to figure out the number of steps at each corner, which is too annoying, or it's the same at each corner, which I previously argued was not adequate.   

It might work to specify an approximate step length.  (I don't really want to have to calculate the bezier length, though.)  Actually the truth is that because the bezier parameterization is nonuniform, really doing this right may be quite tricky.  The bezier parameter may, ideally, need to be sampled in a nonuniform fashion.

revarbat, 2019-03-26:
Well, I can create a `bezier_length()` function that can approximate the curve length to an arbitrary level of precision, if that helps.

adrianVmariano, 2019-03-26:
I think what we really want is a bezier_curve(N) function that returns the bezier sampled into N segments that are of (approximately) equal length.   I guess there's the question of how bad the nonuniformity is for cases of real interest.  For some cases I explored the speed of the parameterization approaches zero...but those are not the cases that deliver a nice curve.  If the nice curves have nice parameterizations, we probably shouldn't work so hard.

adrianVmariano, 2019-03-26:
I checked the behavior of the parameterization of the bezier curves, and found that for middling parameter values, the speed of the curve behaves in a fairly reasonable manner, with the arc length of one tiny step varying by about 50% across the curve.  For a super gradual curve the factor was 4, with larger steps in the flat area, so probably reasonable behavior.  For a really abrupt curve the ratio was 5, but those curves don't look better than circles, so no reason to worry.  

So the conclusion is, sampling the bezier parameter uniformly is good enough.  No need to get fancy.

revarbat, 2019-03-27:
Added some bezier functions that may be useful towards making  the wanted N ~equal length segment bezier routine.  In dev branch: commit 1dc7df18fb1738d56496a3aa4681a3feea11deb2

adrianVmariano, 2019-03-27:
It occurs to me that the real way to sample the bezier would be proportional to its local curvature.

revarbat, 2019-03-28:
Added proper bezier patch/surface support.  Dev branch commit 1d4bf6dcda31f1f369655eb093b464347807b1f2

revarbat, 2019-03-30:
Added bezier triangular patch support.  Dev branch commit b8239d2dde9765cc10070dee84b4e4a03d24d44b

adrianVmariano, 2019-03-30:
Two comments.  Those names are incredibly long.  What's wrong with just "bezier_patch" and "bezier_triangle"?  

Second thing I noticed in testing bezier_triangle_vertices_and_faces is that it's really slow compared to my bezier triangle implementation.  Now to be fair, mine only computes the vertices.  But when I set splinestep to 100 my code ran in 1s (including the hull operation to display the result) and your code took 55 s.

revarbat, 2019-03-30:
This is why I’m using a dev branch now, so these things become apparent before release. 

 Is the slowness in bezier_triangle_point()?  Because I could swear I swiped the basic implementation from one of your emails. Or is the slowness caused by the vertex list mergers and face reindexing?

adrianVmariano, 2019-03-30:
It sure is annoying to do speed testing in OpenSCAD.  My first discovery is that your select() is apparently very slow compared to the simple one I included in my post.  To avoid dependency on BOSL for that thread I posted the simple select:

function select(vector,indices) = [ for (index = indices) vector[index] ];

When I commented this out, my code went from 1s to 15s.   And I also got 5 of those warnings that I thought had been fixed:

WARNING: len() parameter could not be converted, in file lib/BOSL-dev/compat.scad, line 60

It appears that there must be some other path to creating this type of warning, because previously I was getting it from "select([-1,0,1],1)" and in that case, it is fixed.  

It gets weirder, though.  I commented out most of bezier_triangle... and your code is computing the points in 7 s instead of 15s.  I thought maybe it was the for loop structure, but nope.  Also your code doesn't generate the warnings I was getting.  

So I changed my select invocation from select(P[i], 1,len(P[i])-1) to select(P[i],1,-1) and made the similar change on the next line.  Run time went from 15s to 8s.  I changed the third line from [1:n] to 1,n and recovered the last second.  

This is kind of crazy.  Does it have to do with the wrapping?  Or is it error handling that is creating this big hit.  It seems like a serious problem.  

Evidently the rest of the time has to do with assembling the faces.

revarbat, 2019-03-30:
Okay, I was able to get a bunch of speedup by not trying to remove repeated vertex points when joining new patches together.  Apparently polyhedron() doesn't care.  I changed out my select()s in bezier_triangle_point() for list comprehensions and got another doubling of speed.

Overall, I've gone from 12 seconds on my test rounded bezier patch cube, to 3.

revarbat, 2019-03-30:
BTW, my test cube is fully constructed from patches.  It has 8 bezier triangle corners, 12 rectangular bezier patch edges, and 6 rectangular bezier patch faces.

```
include <BOSL/constants.scad>
use <BOSL/transforms.scad>
use <BOSL/beziers.scad>
use <BOSL/math.scad>

function CR_corner(size, orient=[0,0,0], trans=[0,0,0]) =
	let (
		r = 0.4,
		k = r/2,
		// I know this patch is not yet correct for continuous
		// rounding, but it's a first approximation proof of concept.
		// Currently this is a degree 4 triangular patch.
		patch = [
			[[1,1,0], [1,r,0], [1,0,0], [1,0,r], [1,0,1]],
			[[r,1,0], [k,k,0], [k,0,k], [r,0,1]],
			[[0,1,0], [0,k,k], [0,0,1]],
			[[0,1,r], [0,r,1]],
			[[0,1,1]]
		]
	) [for (row=patch)
		translate_points(v=trans,
			rotate_points3d(v=orient,
				scale_points(v=size, row)
			)
		)
	];

function CR_edge(size, orient=[0,0,0], trans=[0,0,0]) =
	let (
		r = 0.4,
		a = -1/2,
		b = -1/4,
		c =  1/4,
		d =  1/2,
		// I know this patch is not yet correct for continuous
		// rounding, but it's a first approximation proof of concept.
		// Currently this is a degree 4 rectangular patch.
		patch = [
			[[1,0,a], [1,0,b], [1,0,0], [1,0,c], [1,0,d]],
			[[r,0,a], [r,0,b], [r,0,0], [r,0,c], [r,0,d]],
			[[0,0,a], [0,0,b], [0,0,0], [0,0,c], [0,0,d]],
			[[0,r,a], [0,r,b], [0,r,0], [0,r,c], [0,r,d]],
			[[0,1,a], [0,1,b], [0,1,0], [0,1,c], [0,1,d]]
		]
	) [for (row=patch)
		translate_points(v=trans,
			rotate_points3d(v=orient,
				scale_points(v=size, row)
			)
		)
	];

module CR_cube(size=[100,100,100], r=10, splinesteps=8, cheat=false)
{
	s = size-2*[r,r,r];
	h = size/2;
	corners = [
		CR_corner([r,r,r], orient=ORIENT_Z,     trans=[-size.x/2, -size.y/2, -size.z/2]),
		CR_corner([r,r,r], orient=ORIENT_Z_90,  trans=[ size.x/2, -size.y/2, -size.z/2]),
		CR_corner([r,r,r], orient=ORIENT_Z_180, trans=[ size.x/2,  size.y/2, -size.z/2]),
		CR_corner([r,r,r], orient=ORIENT_Z_270, trans=[-size.x/2,  size.y/2, -size.z/2]),

		CR_corner([r,r,r], orient=ORIENT_ZNEG,     trans=[ size.x/2, -size.y/2,  size.z/2]),
		CR_corner([r,r,r], orient=ORIENT_ZNEG_90,  trans=[-size.x/2, -size.y/2,  size.z/2]),
		CR_corner([r,r,r], orient=ORIENT_ZNEG_180, trans=[-size.x/2,  size.y/2,  size.z/2]),
		CR_corner([r,r,r], orient=ORIENT_ZNEG_270, trans=[ size.x/2,  size.y/2,  size.z/2])
	];
	edges = [
		CR_edge([r, r, s.x], orient=ORIENT_X,     trans=[   0, -h.y, -h.z]),
		CR_edge([r, r, s.x], orient=ORIENT_X_90,  trans=[   0,  h.y, -h.z]),
		CR_edge([r, r, s.x], orient=ORIENT_X_180, trans=[   0,  h.y,  h.z]),
		CR_edge([r, r, s.x], orient=ORIENT_X_270, trans=[   0, -h.y,  h.z]),

		CR_edge([r, r, s.y], orient=ORIENT_Y,     trans=[ h.x,    0, -h.z]),
		CR_edge([r, r, s.y], orient=ORIENT_Y_90,  trans=[-h.x,    0, -h.z]),
		CR_edge([r, r, s.y], orient=ORIENT_Y_180, trans=[-h.x,    0,  h.z]),
		CR_edge([r, r, s.y], orient=ORIENT_Y_270, trans=[ h.x,    0,  h.z]),

		CR_edge([r, r, s.z], orient=ORIENT_Z,     trans=[-h.x, -h.y,    0]),
		CR_edge([r, r, s.z], orient=ORIENT_Z_90,  trans=[ h.x, -h.y,    0]),
		CR_edge([r, r, s.z], orient=ORIENT_Z_180, trans=[ h.x,  h.y,    0]),
		CR_edge([r, r, s.z], orient=ORIENT_Z_270, trans=[-h.x,  h.y,    0])
	];
	faces = [
		// Yes, these are degree 1 bezier patches.  That means just the four corner points.
		// Since these are flat, it doesn't matter what degree they are, and this will reduce calculation overhead.
		bezier_patch_flat([s.y, s.z], N=1, orient=ORIENT_X,    trans=[ h.x,    0,    0]),
		bezier_patch_flat([s.y, s.z], N=1, orient=ORIENT_XNEG, trans=[-h.x,    0,    0]),

		bezier_patch_flat([s.x, s.z], N=1, orient=ORIENT_Y,    trans=[   0,  h.y,    0]),
		bezier_patch_flat([s.x, s.z], N=1, orient=ORIENT_YNEG, trans=[   0, -h.y,    0]),

		bezier_patch_flat([s.x, s.y], N=1, orient=ORIENT_Z,    trans=[   0,    0,  h.z]),
		bezier_patch_flat([s.x, s.y], N=1, orient=ORIENT_ZNEG, trans=[   0,    0, -h.z])
	];
	// Generating all the patches above took about 0.05 secs.

	if (cheat) {
		// Generating the points for the corners takes 5 seconds on my weak-sauce laptop.
		// Hulling it takes less than a second.
		hull() bezier_polyhedron(tris=corners, splinesteps=splinesteps);
	} else {
		// Generating the polyhedron fully from bezier patches takes 12 seconds on my laptop.
		bezier_polyhedron(patches=concat(edges, faces), tris=corners, splinesteps=splinesteps);
	}
}

CR_cube(size=[100,100,100], r=20, splinesteps=16, cheat=false);
cube(1); // To make CGAL actually have to work.
```

revarbat, 2019-03-30:
Argh.  No, CGAL cares about duplicate vertexes after all.  Gonna have to figure something out.

adrianVmariano, 2019-03-30:
It seems like we'd like the select() function to be usable in general, not something you need to work around when you need decent performance.  I'm investigating this...

Weird that duplicate vertices are a problem.  The manual claims that they are OK here:

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron

Do we need to edit the manual?

revarbat, 2019-03-30:
Well, I get this on trying to render:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330
```

revarbat, 2019-03-30:
Orrrrr, I could just have the faces on my corner patches reversed.

adrianVmariano, 2019-03-31:
I did a bunch of speed testing and kept cutting code out of your select() definition.  This reduced run time from 7s to 5s.  Then I copied your definition into my source file and it ran in 1s.  This seems kind of troubling.  

Finally I copied your select() along with posmod() and modrange() from math.scad into my file.  Run time rose to 24s.  Very confused here.  

The example polyhedron from the manual doesn't produce an error like that...but it does seem to have a problem.  One of the faces is not there and it gets the "may not be a 2-manifold" error.

revarbat, 2019-03-31:
Okay, verified that `polyhedron()` is fine without duplicate vertex culling.

revarbat, 2019-03-31:
Apparently, `posmod()`, and thusly `modrange()` as well, are expensive operations to put at the inner layer of a loop.  I've rewritten `select()` slightly to vastly optimize its speed for the `select(list,start,end)` branch where end<start.

revarbat, 2019-03-31:
Hmmmmmmmmmm.

This takes 47 seconds:
```
function posmod(x,m) = (x%m+m)%m;
foo=[for (i=[1:9999], j=[1:999]) posmod(i,j)];
```

This takes 40 seconds:
```
function posmod(x,m) = x+m;
foo=[for (i=[1:9999], j=[1:999]) posmod(i,j)];
```

And this takes 15 seconds:
```
foo=[for (i=[1:9999], j=[1:999]) (i%j+j)%j];
```

Function calls have high overhead?
The number of defined functions doesn't seem to matter.

revarbat, 2019-03-31:
Just to make sure theres no overhead from making the list with the comprehension, I did this, and it took 53 seconds:
```
function p(x,m) = (x%m+m)%m;
for (i=[1:9999], j=[1:999]) {q=p(i,j);}
```

revarbat, 2019-03-31:
Un-optimized tail recursion: 37 seconds:
```
function p(i) = i<=0? i : p(i-1)+0;
for (i=[1:9999]) {q = p(999);}
```

Optimized tail recursion: 12 seconds:
```
function p(i) = i<=0? i : p(i-1);
for (i=[1:9999]) {q = p(999);}
```

List comprehension: 4 seconds:
```
function p(i) = [for (j=[0:i]) j];
for (i=[1:9999]) {q = p(999);}
```

revarbat, 2019-03-31:
With the latest optimizations in place, my fully bezier patch cube is now down to about 1.5 seconds.

adrianVmariano, 2019-03-31:
I forgot to click the comment button when I wrote here about my findings last night where the ident3 and ident4 definitions were leading to a big slowdown, but it looks like you saw my forum post.  It seems bizarre that nobody saw this issue as a serious problem and it languished for 5 years.  

If they don't fix it my polyhedra library will need to be included rather than used if you plan on calling it many times:  running the demo takes 6s from the library itself and 55s if you "use" the file due to recalculating the library data structure, I assume.

adrianVmariano, 2019-03-31:
How many spline steps did you use for your cube test at 1.5s?   When I did my 3d printed roundcorners tests I did the first print with 20 and the facets were visible and detracted from the smoothness I was going for.  I increased to 200 and got a good result on my send print job.

revarbat, 2019-03-31:
16.  You realize that a splinestep setting of 200 makes for 40000 vertices per rectangular patch, and about half that per triangular patch. For the patch cube that means about 880,000 vertices. 

In 3D printing, most printers can only handle printing a couple vertices per mm before GCode buffer underflows start causing motion stuttering that causes blobby corners. 16 steps on the corners is kind of pushing it as is.

revarbat, 2019-03-31:
The bezier patch code has been merged into master, but the actual patch forms for this are still unknown.  Those, and the above linear code should probably go in their own library file.

revarbat, 2019-04-01:
I've written thousands of lines of OpenSCAD code over the years.  The more weirdnesses like the use vs include issue I see, the more I'm starting to think of OpenSCAD like Postscript:  It may be a Turing complete language, but humans should never be exposed to it.  OpenSCAD should probably be generated from abstractions in a real language, and hidden completely from view.

adrianVmariano, 2019-04-08:
Going back to the question of step size, I printed test models that were 50 mm squares and rounded the corners.  When I did this with 20 steps **it was not enough**.  My print produced distinct facets that basically concealed the effect I was investigating.  The facets were over 1mm in size, I think.  I had to reprint with more steps.  When I changed to 200 I got a nice, continuous result.   Keep in mind that the corner is pretty large.  Since the object is 50 mm on a side we have a corner length of about 50mm, so choosing 200 steps would be expected to give a facet size of about 0.25 mm.   On the circular curvature test print I don't see any systematic defects, even though it has a might higher point density than this.  One the most gradual continuous curvature print where the rounded corner covers the entire side of the square, I think I still see facets, which makes me wonder if the print might benefit from more than 200 points.  This might also be an issue of uniformity of the parameterization.  Another point: my printer is the Prusa Mk3, which is FDM.  Someone printing with SLS, which has higher resolution, might find more points desirable.  

I would think that the effect of "too many" vertices would depend on the intelligence of the slicer.  In other words, does it mindlessly put it all into the gcode, or does it do some sort of integration when the number of points is excessive.  I've generally tended to err on the side of excess points and not noticed any problems with print quality that seem related to this.  

Of course, things may be different in the vertical direction, where layer thickness is going to play a big role.   I feel like I need to print my test pieces sideways and see if the results are different.   I checked the specs for the Prusa SL1 and the minimum layer thickness is 0.01mm.   My printer goes down to 0.05mm, which suggests a potential use of 20 points per mm in the vertical direction.  I usually print at .15mm layers.

revarbat, 2019-04-12:
As the bezier patches and triangles are in and working, and the exact meshes needed for continuous rounding aren't quite worked out, I'm going to defer adding the CR implementations until BOSL2.

adrianVmariano, 2019-04-13:
I haven't really been thinking hard about the 3d part of this.  It seems the way to go is a standard bezier patch with two corners identified together.  

As a sort of companion to roundcorners I've written pathoffset() and another module I'm calling swath() but I don't like the name.  The first one takes a path and shifts it (in the normal direction) by the specified offset---like the built in offset but for paths.  The second function takes a path and expands it into a polygon bounded by two parallel lines.  I'm struggling to figure out end treatments for the second function.  And looking for a better name.  

Is there a sweep function that will assemble a sequence of polygons into a polyhedron?  I also want (actually for immediate use) a rounded box command that will take the output of shell and make an object that is rounded in 3d.  

Also, what do you think about having modules and functions with the same name, e.g. a swath() function that returns a path and a swath() module that returns an object?

revarbat, 2020-04-19:
Deferred to BOSL2, where this is already implemented.

closed 2020-04-19 by revarbat