revarbat/BOSL #20: assertion() doesn't really work

adrianVmariano, 2019-03-26:
The assertion() function is using a version test that tests false on my OpenSCAD, even though my OpenSCAD supports assert.  This enabled me to observe the behavior on legacy versions, which is that assertion() does nothing, at least in functions.   It tests version, and then decides it can't run assert() so it calls f_echo() instead, but f_echo tests version and runs only if the same version threshold is met, which fails, so then nothing happens.  

Also I think that assert_in_list should be rewritten like this:

function check_in_list(argname, val, l, idx=undef) =
	let (success = in_list(val,l, idx))
	success ? true : let(
		msg = str(
			"In argument '", argname, "', ",
			(is_str(val)? str("\"", val, "\"") : val),
			" must be one of ",
			(is_def(idx)? [for (v=l) v[idx]] : l)))
                echo(str("<span style=\"background-color: #ffb0b0\"><b>", "ERROR", ":</b> ", msg, " </span>"))
               false;

and then you call it like:

    assert(check_in_list("parm",parm, ["abc","def","ghi"]),"")

This doesn't address the backwards compatibility issue, which seems impossible if there's no way to halt execution and no way to print a message from a function.  I assume you're not going to try to maintain backwards compatibility once the 2019 reversion is officially released? 
 
Also note that no matter what you do, the color you're using for error output is wrong.  It should be ffb0b0 to match the standard error color.    Your color is a much darker red and I can barely read the text.  Also the use of <p> tags makes the color extend too far and it looks funny.  

The warning color is ffffb0, which you also have wrong, it looks like.

revarbat, 2019-03-27:
What does your version of OpenSCAD output for:

```
echo(version_num());
```

revarbat, 2019-03-27:
Pushed fixes for colors and version checks to master in commit 0fa53eee96c932cbe21f6eedc60ea14a0ec3a6b5

adrianVmariano, 2019-03-27:
If I run this:

    echo(version_num());
    echo(version());

I get this:

    ECHO: 2.01901e+7
    ECHO: [2019, 1, 0]

Is there some way to get it to print more digits of a number?  I couldn't find it.  

But the observation is that suppose you find the right version number where assert was added.  You can change the version number check in the code, and then assertion() will call assert() for me.  But for someone who doesn't have assert(), the assertion() function does nothing.  It doesn't print a message.  It doesn't halt.  And it doesn't seem like you can fix that.  Was echo in functions added at an earlier version than assert?  Basically you can just write

function assertion(cond, msg) = version_num() > something ? assert(cond,msg) : 0;

and you'll get the same behavior.

revarbat, 2019-03-27:
As it turns out, that's more or less how the function assertion() is defined now.

revarbat, 2019-03-27:
The point of it being, that you write the code once, and if it can do assert() on your version of OpenSCAD() (which version is still being determined), then it will.  Otherwise it won't error out or throw warnings about nonexistent assert() functions.

adrianVmariano, 2019-03-27:
Yeah, but for someone reading the code assertion() has a whole lot of extraneous stuff going on that ends up doing nothing so it's kind of misleading.   I was expecting *something* to happen.   If you write it like I posted above then at least it's obvious that nothing is going to happen.  

Another option would be to make a compat.scad file that is used by those running the old version and just define assert() in there.  It doesn't quite provide full compatibility because it would be a function.  Are you planning to try to maintain backward compatibility after the new version comes out?   If you are not then it seems not really worth working too hard on this stuff.   (And I'd argue that so much is missing from the last stable version that it's pointless to try to maintain backward compatibility.)

revarbat, 2019-03-28:
Apparently testing version_num() >= 20190100 isn't reliable.  There are verified releases of OpenSCAD that report themselves as 20190100, which DO support assert(), functional echo(), etc.  AND there are apparently some that report the same version that do NOT support assert(), functional echo(), etc.

Something is rotten in the state of Denmark.

revarbat, 2019-03-28:
Changed tests to >20190100.  Implemented in commit 8a0dc9ebb2dc9380d9019ccd96e23a825468cae7

revarbat, 2019-03-28:
> Are you planning to try to maintain backward compatibility after the new version comes out?

I'll believe in a new official release when I see it.  They've taken over four years since the last one.

closed 2019-03-28 by revarbat