revarbat/BOSL #21: [BUG] Bogus warning message from select()

adrianVmariano, 2019-03-29:
echo(select([-1,1,-1],1));

Produces the warning:

WARNING: len() parameter could not be converted, in file lib/BOSL/compat.scad, line 60

adrianVmariano, 2019-03-29:
I also get 

WARNING: len() parameter could not be converted, in file lib/BOSL/compat.scad, line 75

not sure from what source.  

Line 60:
function is_str(v) = (version_num()>=20190303)? is_string(v) : (is_def(v) && is_def(len(v)) && (len(str(v,v)) == len(v)*2));

Line 75:
function is_array(v) = (version_num()>=20190303)? is_list(v) : (!is_str(v) && is_def(len(v)));

revarbat, 2019-03-29:
I cannot seem to reproduce this warning.

revarbat, 2019-03-29:
Hmm.  The code you quoted does not match my code.  Can you fetch the dev branch and try again with it?

revarbat, 2019-03-29:
My is_def, is_boolean, is_scalar, is_str, and is_array all return the correct results for all value types under both OpenSCAD 2019-03-03 and 2015-03-03

adrianVmariano, 2019-03-29:
My version returns 2019 01 0, so it fails all the version tests, which require that the version be larger than this.  It appears that is_str() is causing an error message because it calls is_def(len(v)) and for v=1, this prints the above error.

revarbat, 2019-03-29:
Really starting to get aggravated at the OpenSCAD developers completely botching their release versioning.  I can't change the version test to accept 20190100, because they released an RC2 that reports that version, that doesn't support is_* calls.  Yet they released another RC3 that doesn't have is_* calls, but still reports 20190100.  AND it throws warnings about len(scalar) now.

And I can't test that version, without having to jump through compilation hoops, because it wasn't released on OS X.

I have an idea, though...  Try this definition of is_str:

```
function is_str(v) = (version_num() > 20190100)? is_string(v) : (v=="" || (is_def(v) && is_def(v[0]) && (len(str(v,v)) == len(v)*2)));
```

revarbat, 2019-03-29:
Should be fixed in master branch commit 57adf91b65f81cf189c69b4849b46849593412b3

adrianVmariano, 2019-03-30:
I tried the previous code that produced the warning and it's no longer doing it.

closed 2019-03-30 by revarbat