revarbat/BOSL #26: 2d mutators

adrianVmariano, 2019-03-31:
I think ultimately we need a whole new slew of 2d stuff, but at the moment I'm annoyed because the mutators are 3d, but I need to clip my object before doing other 2d operations, so I can't extrude into 3d yet.  In my own version of this I implemented half-spaces that I could rotate and I included an argument (plane=true) to get half-planes instead of half-spaces.  So that's one approach.  Alternatively there could be right_half2d() and so on.

revarbat, 2019-03-31:
Hmm.  Most of transforms.scad is fine with 2D.  It looks like the main ones that don't work with 2D (other than the obviously inapplicable ones) are the *_half() and skew_xy() modules.  I can see adding an argument to those to get a planar version.   `plane=true` seems reasonable, but I kind of want to use `planar=true` as it's an adjective rather than a noun.

revarbat, 2019-03-31:
I have been thinking of a shapes2d.scad file, for actual 2D shapes like rounded rects, real trapezoids, etc.

revarbat, 2019-03-31:
Added `planar` args to skew_xy() and *_half() modules.  Master branch commit 5a09252c5cd57602ee31a5c0f7c140108477be96

closed 2019-03-31 by revarbat