revarbat/BOSL #27: [BUG] xpread manual error

adrianVmariano, 2019-04-01:
The manual says that p1 defaults to the origin.  Wording is a bit unclear, but it says "the line" starts at p1.  But I was getting my items symmetric around the origin, not starting there.  When I added p1=[0,0,0] they started at the origin, so this behavior is not actually default.  (Also why is "p1" the parameter name.  Non-obvious.  Perhaps this makes sense if I can set p2=[0,0,0] and get the items going to the left of the origin?)

revarbat, 2019-04-01:
I'm afraid `p1` is probably an artifact of having just redone `spread()` which has `p1` and `p2` arguments for start and endpoints.  In `spread()` if you only specify `p1`, and not `p2`, but you give it spacing info, it'll start the line at `p1`.

In any case, I'll tweak the docs.

Also, it's a recent enough change I can probably get away with renaming `p1` to `sp`.

revarbat, 2019-04-02:
Fixed in master branch commit d97fd84de57f30b9fd555a36a7cebea5d92c4016

closed 2019-04-02 by revarbat