revarbat/BOSL #28: Add round2d and shell2d modules

adrianVmariano, 2019-04-01:
```
use<BOSL/math.scad>
use<BOSL/transforms.scad>

// Round an arbitrary 2d object
//
// Specify the rounding radius as r and the function will
// round all convex and concave corners with this radius.  
//
// Specify the outer radius (for convex corners) with or
// Specify the inner radius (for concave corners) with ir
//
// The ir and or parameters override r.  You can set either
// one to zero to to round only inside or outside corners.  
//
//
// The 2d object must not have any sections narrower than
// the 2*or (twice the outer radius).  Such sections will
// disappear.  
// 

module round2d(r,or,ir){
  or = is_def(or) ? or :
       is_def(r) ? r : 0;
  ir = is_def(ir) ? ir :
       is_def(r) ? r : 0;
  offset(or){
    offset(-ir-or){
      offset(delta=ir){   // Does delta vs r make any difference?
        children();
      }
    }
  }
}

// thickness can be positive to expand outward, negative to shrink inward,
// or an array to do both.
//
// or is the outside radius, ir is the radius of concave corners on the outside
// (same as for round2d)
//
// fill is a fill radius for filling in deep corners inside the shell
// round is a rounding radius for rounding points inside the shell

module shell2d(thickness, or=0, ir=0, fill=0, round=0)
{
   thickness = is_scalar(thickness) ?
                        (thickness<0 ? [thickness,0] :
                                       [0,thickness])
                      : thickness[0]>thickness[1] ?
                                  [thickness[1],thickness[0]] :
                                  thickness;
   difference(){
     round2d(or=or,ir=ir)
       offset(delta=thickness[1])
         children(0);
     round2d(or=fill,ir=round)
       offset(delta=thickness[0])
         children(0);
   }
}

///////////////////////////////////////////////////////
// Examples

$fn=64;

points=[[-3,0],[5,3],[-1,7],[8,7],[20,20],
        [15,10],[17,0],[12,5],[10,0]]; 

ydistribute(spacing=22){
  polygon(points);
  round2d(r=1)polygon(points);
  round2d(ir=0,or=.5)polygon(points);
  round2d(ir=1)polygon(points);
  round2d(ir=2,or=1) difference() { circle(r=5); square([6,6]);}
  shell2d(.5) polygon(points);
  shell2d(.5,or=.5) polygon(points);
  shell2d(.5,ir=1) polygon(points);
  shell2d([-.5,.5])polygon(points);
  shell2d(.5,fill=.5)polygon(points);
  shell2d(-1,round=1)polygon(points);
}
```

I needed round2d yesterday for an actual real project.   It seems like a commonly written function.  I'm not sure where they should go.  I was thinking this could go in a rounding.scad file along with roundcorners and the hopeful smooth rounded cube that still isn't working right.    Alternatively maybe we want a file of 2d functions and they belong there?

revarbat, 2019-04-02:
I like these, but I really didn't understand the args `ir`/`or`/`fill`/`round` until I ran the demo.
I'd suggest the names `oround`, `ofillet`, `iround`, `ifillet`.   Though my previous uses of the term `fillet` muddies the waters.

revarbat, 2019-04-02:
I'm of mixed minds about whether to start a rounding.scad with these, or if they should be put in a 2D Mutators section of transforms.scad.

revarbat, 2019-04-02:
Those instances of `children(0);` are definitely supposed to be `children();`.

adrianVmariano, 2019-04-02:
Yeah, you're right about the children.  The code I was inspired by did something weird with the rest of the children. 

I struggled to come up with good names for those parameters and the code I was inspired by had parameters whose very function was actually mystifying.  I like your suggestion of [io]{round,fillet}.  If you're planning to switch to using fillet for internal rounding and round for external I think it will be OK.  

I'm not sure there's enough rounding stuff to justify a rounding.scad.  I was just trying to think it through, and this is what I've got in mind that could potentially go there

* round2d
* shell2d
* roundcorners
* super smooth cube

It seems like this might not be enough to warrant its own file.   I was also thinking about a box function that could create a box (meaning a container with thin walls and an open top) with rounded edges (sort of like the 3d version of shell2d) would be nice, but I'm not sure how to write that function.  It seems like it would need to be done based on a path and constructing a polygon.  

I think we need some other 2d stuff to make both paths and geometry, so maybe you want a shapes2d.scad?   What is the difference between a path and a polyline, by the way?

revarbat, 2019-04-02:
Another case of my using words interchangeably.  Polyline is a term I've seen used a lot with CADs, and I used it in my own CAD.  So sometimes when I'm working on paths code, I'm thinking polyline.

revarbat, 2019-04-03:
Added to transforms.scad.  Implemented in master branch commit b86206a80cfca4f8c6a51b84794ec3d7b4d94137

closed 2019-04-03 by revarbat