revarbat/BOSL #32: Add attachment scheme.

revarbat, 2019-04-05:
BOSL should implement some sort of attachments scheme for assembling parts together.

Other implementation references:
- https://github.com/CarlosGS/tutorial-OpenScad/blob/master/ejemplos_openscad/utils/obiscad/attach.scad
- http://www.iearobotics.com/blog/2012/09/10/enhancing-openscad-with-the-attach-library/

adrianVmariano, 2019-04-05:
Two other things to consider for ideas related to this are the local library and the General Library of Relativity. 

* https://github.com/jreinhardt/local-scad
* https://github.com/davidson16807/relativity.scad

revarbat, 2019-04-07:
Oooooh.  There's some particularly interesting ideas in the relativity library, that are NOT incompatible with the current BOSL library..

revarbat, 2019-04-09:
A lot of changes needed in my library, but I can make just about anything that uses orient_and_align() allow for attachable children by named attachment points.

revarbat, 2019-04-12:
I have a reasonably good implementation of this, but I'm going to defer it until the BOSL2 cleanup, as there are several things that can be speed optimized if I can break backwards compatability.

revarbat, 2019-04-17:
Changing `use` to `include` lowers runtime to 15 seconds.

revarbat, 2019-04-25:
Implemented in BOSL2:  https://github.com/revarbat/BOSL2

closed 2019-04-25 by revarbat