revarbat/BOSL #37: Interior fillet around a cylinder

sej7278, 2019-04-18:
How could we make a fillet around the base of a cylinder (like a tree stump) to provide lateral support for a 3D printed pillar? Example image from wikipedia:

[image]

A similar sort of thing has been asked on the OpenSCAD forum, but seems to be for arbitrary shapes: http://forum.openscad.org/Fillet-With-Internal-Angles-td17201.html

cyl() has chamfers and fillets but they go the wrong way - I did try ```fillet1=-4``` !

revarbat, 2019-04-18:
Oddly enough, I believe that fillet_hole_mask() would fill this specific purpose.

sej7278, 2019-04-18:
I came up with this, but don't want to see the cylinder at the bottom:

```
$fn=100;
rotate([180,0,0])
{
    cylinder(h=12, d=73, center=true);
    fillet_hole_mask(r=73/2, fillet=10);
}
```

revarbat, 2019-04-18:
Okay, I've added an `overage` argument to both `chamfer_hole_mask()` and `fillet_hole_mask()`.
Committed to master branch in commit 46e668548c33166fdf56ef49533328fd1bd04f6a

sej7278, 2019-04-19:
Oh that's excellent thanks, some example code to make a lovely pillar/spacer:

```
include <BOSL/constants.scad>
use <BOSL/masks.scad>
use <BOSL/transforms.scad>
use <BOSL/shapes.scad>

$fn = 100; // smoothing factor
wall = 2; // thickness of cylinder wall
over = 0.5; // height of base of fillet
fill = 5; // fillet radius
radius = 20; // cylinder radius
length = 50; // cylinder height

difference()
{
    union()
    {
        cyl(l=length, r=radius, center=false);
        yrot(180) down(over) fillet_hole_mask(r=radius, fillet=fill, overage=over);
    }
    
    up(wall) cyl(l=length, r=radius-wall, center=false);
}
```

closed 2019-04-19 by sej7278