revarbat/BOSL #38: simplify_path not getting tail recursion?

adrianVmariano, 2019-04-20:
I tried to use simplify_path on a long list and it failed with a recursion error.  It looks to me like it should be getting tail recursion, so not sure what's going on there.

revarbat, 2019-04-21:
I specifically designed that function to enable tail recursion optimization.  WTH?  Not sure what to do about this.

adrianVmariano, 2019-04-21:
Maybe worth posting it to the forum and asking why it's not getting tail recursion?  It looks to me like it fits the pattern and should get tail recursion, and we need to understand why it's not.  Could it be an OpenSCAD bug.  

I think this can be written non-recursively if you first compute a boolean vector that indicates colinearity and then write a for statement that outputs a given point only if one of the points on either side is not collinear.   I'm not sure how run time would compare between this approach and the recursive approach (if tail recursion were working), but I recall in your testing plain list comprehensions were significantly faster than tail recursion, so it might even be better.

revarbat, 2019-05-02:
Reimplemented using list comprehensions.

closed 2019-05-02 by revarbat