revarbat/BOSL #39: convex_hull() name confusing?

adrianVmariano, 2019-04-20:
Ok, so what's the difference between hull() and convex_hull()?  From a user perspective, I think the name "convex_hull()" doesn't at all differentiate it.  

The Linde functions return the boundary of the hull.  In 3d, the face index list, in 2d, the edge index list.  Since we already have "hull" adding the word "convex" doesn't really give any clarity.  So I wonder if 

hull_boundary,  hull3d_faces, and hull2d_edges might be good names for these functions.

sej7278, 2019-04-20:
got to agree with you here @adrianVmariano i've no idea what this does and there's no example code or image

revarbat, 2019-04-25:
My proposed renamings are `hull2d_path()` and `hull3d_faces()`.  Example new docs for them are already in the pre-alpha BOSL2: https://github.com/revarbat/BOSL2/wiki/hull.scad

revarbat, 2019-04-25:
I can backport to BOSL v1 if this looks good.

revarbat, 2020-03-07:
Not back-porting.

closed 2020-03-07 by revarbat