revarbat/BOSL #40: Add hull_points module

adrianVmariano, 2019-04-20:
This trick is very fast and works on huge point sets.  It's also very non-obvious, so I think well worth including.   In the past I have seen it produce warnings from degenerate input sets (but still produce correct results).   I can't duplicate that behavior at the moment.  

```
module hull_points(points){
    extra = len(points)%3;
    list = concat(
                [[for(i=[0:extra+2])i]],
                [for(i=[extra+3:3:len(points)-3])[i,i+1,i+2]]);
    hull() polyhedron(points, faces=list);
}
```

revarbat, 2019-04-21:
I bet if you tested for collinear() on the three points, you could get rid of the degenerate faces.

adrianVmariano, 2019-04-21:
Yeah, I was originally calling simplify_path() to do just that.  But when I tested yesterday I couldn't cause the warning message to appear, even when I fed it a string of colinear points.  So I'm not sure what condition exactly leads to that error message.

revarbat, 2019-04-25:
I've added `hull2d_points_fast()` and `hull3d_points_fast()` to `hull.scad` in BOSL2 already.  I can backport to BOSL v1.0 if these looks fine.

BOSL2 hull.scad docs: https://github.com/revarbat/BOSL2/wiki/hull.scad

revarbat, 2019-04-26:
This is covered by the changed for #39

closed 2019-04-26 by revarbat