revarbat/BOSL #42: Question: how to add thread to non-nut object

sej7278, 2019-04-24:
Hi,

just playing with threading.scad and figured out how to add a thread to the top of a cylinder (although its kind of hovering, could do with filleting against the inside of the cylinder) using ```threaded_rod()``` but can't figure out how to add the opposite thread to the inside of the lid, which I don't want to be a ```threaded_nut()```.

Any ideas - or is this not possible?

I had previously been using [openscad-threads](http://dkprojects.net/openscad-threads/) which uses an ```internal``` flag to provide the opposite thread - assume its not just calling ```difference()```.

[image]

revarbat, 2019-04-25:
In this case, just take the lid and difference away a `threaded_rod()` with a diameter increased by `2*PRINTER_SLOP`

revarbat, 2019-04-25:
Also, the filleting can be made with `fillet_hole_mask()`

sej7278, 2019-04-25:
ok thanks, i think i've got it now, about to print the [new revision](https://gist.github.com/sej7278/5154113b346d7a79173840580fe309d4) to see if the threads go together

closed 2019-04-25 by sej7278

sej7278, 2019-04-25:
Finished the design - test printed fine, https://www.thingiverse.com/thing:3587719