revarbat/BOSL #43: screw threading database

adrianVmariano, 2019-04-26:
Maybe I'm missing something, but it doesn't appear that the screw threading code has a database to generate standard screw threads.  So if I want to make a USA 8-32 (UNC) I have to look up the parameters.  Did I miss something?  

There's been talk of a library that generates screw threads by name which seems desirable.  In other words, to be able to specify "8-32" or "M4" or "M4-fine".

revarbat, 2019-04-26:
SAE bolts, screws, etc are just something I never got around to making a library file for.  I intended to have an `sae_screws.scad`, to correspond to `metric_screws.scad`, but since all my design work was with metric parts, I just never got around to it.  Wanna take care of that? 😄

revarbat, 2019-04-26:
To be honest, I'm a bit unclear still on the difference between SAE and UNC standards, etc.

Surprisingly, I do live in the USA.  I just mostly work metric in 3D printing.

adrianVmariano, 2019-04-26:
https://en.wikipedia.org/wiki/Screw_thread#History_of_standardization

SAE was replaced by the UTS in 1949.  It sounds like lingering use of "SAE" is incorrect labeling of stuff that is actually UTS.  

Almost all of the machine screws I have on hand are UTS of some sort, so if I was designing to use my existing hardware that would be more convenient.  

I didn't notice the separate file metric_screws.scad when I was reading the threading.scad docs.

sej7278, 2019-08-26:
i find this little change to threading.scad useful - you can specify the number of sides to a nut, so instead of the default of 6, you can use 100 for example and get a smooth cylinder

```diff
diff --git a/threading.scad b/threading.scad
index 6669915..5746077 100644
--- a/threading.scad
+++ b/threading.scad
@@ -275,6 +275,7 @@ module trapezoidal_threaded_nut(
        profile=undef,
        left_handed=false,
        starts=1,
+    sides=6,
        bevel=true,
        slop=undef,
        orient=ORIENT_Z,
@@ -284,7 +285,7 @@ module trapezoidal_threaded_nut(
        slop = default(slop, PRINTER_SLOP);
        orient_and_align([od/cos(30),od,h], orient, align) {
                difference() {
-                       cylinder(d=od/cos(30), h=h, center=true, $fn=6);
+                       cylinder(d=od/cos(30), h=h, center=true, $fn=sides);
                        trapezoidal_threaded_rod(
                                d=id,
                                l=h+1,
```

call using:

```python
trapezoidal_threaded_nut(id=24.73, od=27, h=10, pitch=3.18, align=V_TOP, sides=100);
```

closed 2020-04-19 by revarbat