revarbat/BOSL #44: threads for differencing?

adrianVmariano, 2019-04-26:
It appears that the code doesn't let you create threads to subtract from a model.  In other words, if I want a threaded hole in my model I have to make an oversized hole and union in a nut.  That seems sort of clumsy.  (I assume that the threads in a nut are not the same as the threads in a rod---that is, I can't just subtract the rod from my model.)

sej7278, 2019-04-26:
have a look at #42

adrianVmariano, 2019-04-26:
It does not appear that you can turn an external thread into an internal thread by scaling, because the way the threads are truncated is different:

https://en.wikipedia.org/wiki/Unified_Thread_Standard#/media/File:ISO_and_UTS_Thread_Dimensions.svg

So that suggests that the method you suggested in #42 is not really accurate.  

As a separate matter, it's better for reading code and writing it if we have a clear indication that this is an internal thread and this is an external one rather than jiggered radii.  Keep the printer slop parameter for use as actually defining the printer slop.  Presumably in one case it needs to be positive and the other case negative.

sej7278, 2019-04-26:
i didn't suggest the method in #42 but it did work - with some playing around with moving up and down, its far from ideal though. i prefer the [openscad-threads](http://dkprojects.net/openscad-threads/) way of doing things.

revarbat, 2019-04-26:
To be honest, threading.scad doesn't yet truncate the peaks and valleys of threads differently like the UTS spec defines.  An interior thread mask is currently just a radius increase of PRINTER_SLOP over a matching exterior thread, for standard 30º thread angle.

revarbat, 2019-04-26:
For future `threading.scad` improvements, here's some links crib ideas from.

- http://dkprojects.net/openscad-threads/
- https://github.com/MisterHW/IoP-satellite/blob/master/OpenSCAD%20bottle%20threads/thread_profile.scad
- https://github.com/adrianschlatter/threadlib

adrianVmariano, 2019-04-26:
The UTS spec and ISO metric spec are the same with regards to peak/valley truncation.  And also they show the valleys as having a fillet, which would be stronger and presumably especially desirable if someone is 3d printing a screw thread.   The diagram on wikipedia gives the thread diameters with respect to the flats at the valleys and peaks, so if you aren't following the spec then....what are you doing?  It presumably means the threading has the wrong diameter somewhere, no?  

Is square thread even printable?  I was thinking that the buttress thread looks like a really nice 3d print option since it can be oriented so that all the slopes are above 45 deg.

revarbat, 2019-04-27:
I originally wrote threading.scad to make leadscrews, where the thread profile is fully symmetric.  The ISO and UTS thread profiles didn't fit this criterion, so I got it as close as I could, but it wasn't accurate.  I've just coded up a way to pass a thread profile to `trapezoidal_threaded_rod()`, and that will let `threaded_rod()` pass in the asymmetric UTS/ISO triangular thread profiles.  As a side benefit, it'll allow rounding the troughs (or peaks for internal thread masks), and enable things like buttress profiles.

revarbat, 2019-04-27:
Regarding the printability of square (or ACME, or metric-trapezoidal) threads, you can make interior square threads by printing with the threading on its side, if you clip the overhanging top threads with >45 degree overhangs.  I've done this before to make parts that are driven by lead screws.  You can print exterior threaded parts in halves that are assembled post-printing with guides.  If you want to print it in one piece, then you get limited to a `thread_angle` of 40 degrees or so, or to the soon to be added buttress threads.  In my own printed lifter-rod lead screws for the Snappy printer, I used the former method.

revarbat, 2019-04-27:
What threading.scad has been doing up until now for UTS/ISO triangular threads, is to truncate BOTH the top and bottom of the threads at H/4, whereas the spec truncates peaks at H/8 and valleys at H/4.  Which works fine if you use the same library for printing your matching threaded parts, but won't actually match REAL threaded hardware.  Which is a problem, yes.  I'm fixing this now.

revarbat, 2019-04-28:
Thread profiles added in BOSLv1 in fe3d3b98f38765dc1f466b6fbf06e923caa250ce and BOSL2 in 2161e1af61e1c6889929e8d7c430fd76fb0d0c61

revarbat, 2019-04-28:
Got the docs updated.

revarbat, 2019-04-28:
Still would be good to implement thread tapering, so keeping this open.

revarbat, 2019-05-03:
Added thread_helix() with thread tapering in BOSL2.

closed 2019-05-03 by revarbat