revarbat/BOSL #45: staggered sphere: why?

adrianVmariano, 2019-04-26:
How is staggered sphere special or advantageous over the built-in sphere?  

Does it have the property (as discussed recently on the forum) that it has full dimensions along the 3 axes for all values of $fn that are divisible by 4?   Because that property seems nice.

revarbat, 2019-04-26:
It always has vertices at the poles, but I'm actually not sure about the meridian X and Y axis vertices.  Built-in sphere pretty much always fails to have Z pole vertices.

The main point of staggered_sphere() is that it provides a smoother, closer approximation of a sphere. The center of the triangular faces should vary from the ideal sphere less than the center of the built-in sphere's quad faces. This different triangulation may also be preferable aesthetically, and in various low-poly circumstances.

I actually originally wrote it as a bit of a lark, then decided there may be various times it'd be useful, and I didn't want to rewrite it again.  So I merged it into the library, in the (at the time) glaringly obvious missing spheroid section that orthogonality suggested should exist.

revarbat, 2019-04-26:
Empirical testing shows that `staggered_sphere()` has vertices at all axis extremes for `quantup($fn,8)`, or every other `quant($fn,4)`.

revarbat, 2019-04-26:
It looks like I can tweak it to always have axis extrema vertices for every `quant($fn,4)` if I just check for odd multiples, and add a rotation of `180/$fn` for every other quant4.

revarbat, 2019-04-26:
Added to BOSLv1 in master branch commit 9fb74b662002887af9964a9dd3ddb457818bdfe2
Added to BOSL2 in master branch commit 2f290ca1085ee513a7c66e5ce05f015657aa803b

closed 2019-04-26 by revarbat