revarbat/BOSL #49: beveled gears working correctly?

adrianVmariano, 2019-06-03:
I don't really understand gears well, so this may be a false alarm.  But I noticed that when you set bevelang for a gear the diameter of the gear shrinks, possibly by a significant fraction of a tooth.  Is that how it is supposed to work?   Is this for gears that meet at an angle?  (The docs could be expanded somewhat.  Also the documentation is in the wrong order: the gear() module should be documented first, followed by rack, then gear2d, and then the functions last.  That's the order that users are likely to care about things.)  I also found that if the bevel angle got above about 30 then the code starts making double beveled teeth, which seems wrong.  So it should probably either be trapped as invalid user input or fixed, if it's supposed to work.

revarbat, 2019-06-04:
The discrepancy of gear diameter is partly due to the tooth profile lifting up from being perpendicular to the gear axis.  However, it can also be considered a failure in how the pitch is defined for beveled gears.  Do you count the inner pitch, or the outer pitch, or the average pitch for a beveled gear?

revarbat, 2019-06-04:
Reading up a bit on beveled gears indicates that my implementation in `gear()` is naive and incorrect, as beveled gears should have octoid tooth profiles instead of involute profiles.  This indicated that I need a specific `bevel_gear()` module.

revarbat, 2019-06-04:
- https://www.geartechnology.com/articles/0315/A_Practical_Approach_for_Modeling_a_Bevel_Gear/
- https://www.sdp-si.com/resources/elements-of-metric-gear-technology/page4.php

revarbat, 2019-06-04:
Not sure, but may need something for crown gears and herringbone helical gears.

adrianVmariano, 2019-06-04:
https://www.thingiverse.com/thing:1604369

Implements a lot of gears.

revarbat, 2019-06-04:
Hmm.  Octoid profile teeth are the shape you get when you project involute gear teeth onto a spherical surface.  Which is *really* close to the shape you get when you tilt the teeth up to be perpendicular to the bevel angle like I do.  So, not sure if I need to do anything here.

revarbat, 2019-06-04:
I suppose I need to print out a differential gear set or something, and test.

revarbat, 2019-06-04:
Helical beveled gears are *definitely* malformed, though.

adrianVmariano, 2019-06-04:
I feel like I read that printed gears tend not to hold up well, which makes it seem like the herringbone helical gears are probably the best form.  I was wondering if the helical gears were correct.  

Heh.  I was just trying to use the library to solve a problem someone posted to the forum and: can of worms, opened.

revarbat, 2019-06-12:
Just committed new `bevel_gear()` code, that could still be improved, but is far better than previous.  Docs still need work.  Added function version of `gear2d()` and `gear_tooth_profile()`.

closed 2020-03-07 by revarbat