revarbat/BOSL #60: [BUG]

pruby, 2020-04-18:
**Describe the bug**
The 2d involute gear profile with these particular settings seems to have some internal edges, which interfere with extruding it. The following code illustrates the problem, resulting in CGAL errors in OpenSCAD 2019.05 . BOSL has been updated to the current master branch (commit 5af1a8d).

**Code To Reproduce Bug**
```
include <3rdparty/BOSL/constants.scad>;
use <3rdparty/BOSL/involute_gears.scad>;

cylinder(5, d=3, $fn=32);
linear_extrude(5) {
    gear2d(3, 24);
}
```

**Expected behavior**
The resulting 2d gear profile should be a single curve, able to be consistently extruded.

**Screenshots**
Full render (F6) with the whole code above triggers a CGAL error, which results in the gear not being included in the result:

[image]

Full render (F6) of just the gear2d profile shows internal edges are present in a radiating pattern:

[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2019.05
- Other libraries used: none

revarbat, 2020-04-19:
Oh boy.  I went to debug this, and most the useful tools are in BOSL2.  The `gear2d()` code got a major rewrite, and I'll have to see if those can be back-ported to BOSL1.

revarbat, 2020-04-19:
Well, I didn't do a full backport, but I redid how `gear_tooth_profile()` and `gear2d()` generates gears so that it shouldn't ever have cleavages like that.  Fixed in 03bb2712867c57d349f954ed7e7b5925574a4d32

closed 2020-04-19 by revarbat