revarbat/BOSL #64: [BUG] trying to merge half_joiner2 with object cannot render

benjamind2330, 2020-06-08:
**Describe the bug**
You can render in openscad the half_joiner2, i have even exported to stl and printed one. But when attempting to combine it with other things, you cannot render it.

You get the error: 
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426

**Code To Reproduce Bug**
```
include <BOSL/constants.scad>
use <BOSL/transforms.scad>
use <BOSL/joiners.scad>
xrot(-90) half_joiner2(h = 20, w = 10, l = 4, a=15, orient=ORIENT_Y);
zmove(3+5) cube([40, 40, 10], center=true);
```
Then use F6 to render.

**Expected behavior**
I expect to be able to render this simple combination of joiner and shape. 

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2019.05

revarbat, 2020-06-13:
This can be reproduced with just a `half_joiner()` and a cube:
```openscad
include <BOSL/constants.scad>
use <BOSL/transforms.scad>
use <BOSL/joiners.scad>

up(10) half_joiner(h=20);
cube();
```

As the half joiner is constructed purely from geometry and is NOT assembled using `polyhedron()`, this is some sort of bug in OpenSCAD itself.  I'll see if I can work around it.

revarbat, 2020-06-13:
Fixed this by removing calls to `render()` from joiners.scad.
6f8d489d2cb7c00c42cf236626922ebe73e63ff4

closed 2020-06-13 by revarbat

engelchrisi, 2021-02-24:
Would you mind fixing this also in BOSL2. A first test looked promising.
Thanks