revarbat/BOSL #66: [BUG] Possible version incompatibility / document compatibility

terryspitz, 2020-06-10:
**Describe the bug**
When I cut and pasted a grid2d example from the wiki into openscad 2019.05 I get errors.
```

include <BOSL/constants.scad>
use <BOSL/transforms.scad>

// Makes a grid of hexagon pillars whose tops are all
// angled to reflect light at [0,0,50], if they were shiny.
hexregion = [for (a = [0:60:359.9]) 50.01*[cos(a), sin(a)]];
grid2d(spacing=10, stagger=true, in_poly=hexregion) {
    // Note: You must use for(var=[val]) or let(var=val)
    // to set vars from $pos or other special vars in this scope.
    let (ref_v = (normalize([0,0,50]-point3d($pos)) + V_UP)/2)
        half_of(v=-ref_v, cp=[0,0,5])
            zrot(180/6)
                cylinder(h=20, d=10/cos(180/6)+0.01, $fn=6);
}
```

OUTPUT

```
Parsing design (AST generation)...
Saved backup file: C:/Users/terry/Dropbox/Documents/OpenSCAD/backups/unsaved-backup-MgdJlXHJ.scad
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown function 'normalize', in file ../../Users/terry/Dropbox/Documents/OpenSCAD/libraries/BOSL/examples, line 10.
WARNING: max() parameter could not be converted, in file ../../Users/terry/Dropbox/Documents/OpenSCAD/libraries/BOSL/math.scad, line 96
WARNING: min() parameter could not be converted, in file ../../Users/terry/Dropbox/Documents/OpenSCAD/libraries/BOSL/math.scad, line 96
WARNING: acos() parameter could not be converted, in file ../../Users/terry/Dropbox/Documents/OpenSCAD/libraries/BOSL/math.scad, line 956
ERROR: Assertion 'FAILED' failed: "`from` and `to` should be used together." in file compat.scad, line 199
TRACE: called by 'assert', in file compat.scad, line 199.
TRACE: called by 'if', in file compat.scad, line 196.
TRACE: called by 'assertion', in file transforms.scad, line 279.
TRACE: called by 'if', in file transforms.scad, line 278.
TRACE: called by 'if', in file transforms.scad, line 276.
TRACE: called by 'rot', in file transforms.scad, line 1859.
TRACE: called by 'difference', in file transforms.scad, line 1857.
TRACE: called by 'if', in file transforms.scad, line 1847.
TRACE: called by 'if', in file transforms.scad, line 1845.
TRACE: called by 'half_of', in file transforms.scad, line 1846.
TRACE: called by 'translate', in file transforms.scad, line 1846.
TRACE: called by 'if', in file transforms.scad, line 1845.
Compiling design (CSG Products generation)...
Geometries in cache: 6
Geometry cache size in bytes: 1857000
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 9294208
Compiling design (CSG Products normalization)...
Normalized CSG tree has 0 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

```
**Code To Reproduce Bug**
```

include <BOSL/constants.scad>
use <BOSL/transforms.scad>

// Makes a grid of hexagon pillars whose tops are all
// angled to reflect light at [0,0,50], if they were shiny.
hexregion = [for (a = [0:60:359.9]) 50.01*[cos(a), sin(a)]];
grid2d(spacing=10, stagger=true, in_poly=hexregion) {
    // Note: You must use for(var=[val]) or let(var=val)
    // to set vars from $pos or other special vars in this scope.
    let (ref_v = (normalize([0,0,50]-point3d($pos)) + V_UP)/2)
        half_of(v=-ref_v, cp=[0,0,5])
            zrot(180/6)
                cylinder(h=20, d=10/cos(180/6)+0.01, $fn=6);
}
```

**Expected behavior**
Should run and produce output

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2019.05
- Other libraries used: none

revarbat, 2020-06-12:
Example code is missing a `use <BOSL/math.scad>` line.

revarbat, 2020-06-12:
Fixed example in 985a091521becec275346ea198261dbfa04a8b1e

closed 2020-06-12 by revarbat