revarbat/BOSL #67: [BUG] Cuboid fillet is not applied evenly on all edges.

owo, 2020-06-22:
**Describe the bug**
Cuboid fillets along the side edges are different in size and in number of segments than those of the top and bottom edge rings. This issue occurs in both preview and render views.

*Note: I have only been using OpenSCAD for about a day, so I really don't know if this issue is a bug or a feature. However, I am coming from using Belnder and its bevel tools which do not have this issue.*

**Code To Reproduce Bug**

The following code generates two cubes using the `cuboid` function.
The first is a a large yellow cube with a fillet and the second is a smaller purple cube with no fillet for reference.
Both cubes are centered on the y and z axes.

```
use <BOSL/shapes.scad>

color("yellow") cuboid([20, 20, 20], fillet=3);
color("purple") translate([15, 0, 0]) cuboid([10, 10, 10]);
```

**Expected behavior**
I would expect the fillet be consistent on all edges in terms of size and number of segments.

**Screenshots**
Below are two screenshots of the output of the code above.

This first one illustrates how the fillet size on the top and bottom edge loops is smaller than those on the side edges. It's clear that there's a significant size difference between the left and right margins, and the top and bottom ones. 

[image]

The second one illustrates how the number of segments on the side edges (3) is different than those on the top and bottom edge loops (2).

[image]

Below is a screenshot of applying a bevel modifier to Blender's default cube for reference.
While in Blender you can specify both the size (offset) and number of segments, the bevel is applied consistently along all edges.

[image]

**Additional context**
- OpenSCAD Version: 2019.05
- Other libraries used: None

revarbat, 2020-06-23:
The relevant code that is generating this is more or less:
```openscad
minkowski() {
    cube(isize, center=true);
    sphere(r=fillet, $fn=quantup(segs(fillet),4));
}
```
So, as best as I can tell, the shape of the sphere is at fault.

revarbat, 2020-06-23:
This is fairly annoying to fix in BOSL1.  It's relatively easy to fix in BOSL2, at least, so I did.

owo, 2020-06-23:
Thank you for looking into this. I'll give BOSL2 a try.

revarbat, 2020-06-23:
I thought of a simpler solution for BOSL1.  It won't make corners quite as pretty as Blender, but edges should be consistent now.  Fixed in 0faa92137c64cc01c7f8abe34598ab0e47b70435

closed 2020-06-23 by revarbat