revarbat/BOSL #70: "constants.h" ?

3Dietrich, 2020-07-25:
first: Thank You for this library and it's verbose description! It is very handy and very helpful!

**Is your feature request related to a problem? Please describe.**
I never heard about "ORIENT_ constants from constants.h" before and i have no idea what this means. Should I go into openscad source code and search it for "ORIENT_"..?
**Describe the solution you'd like**
Maybe you did this already but I couldn't find: please, just list all the possible ORIENT_ and similiar possibilities here it the wiki. Once, as a part of Terminology or something like that.

**Describe alternatives you've considered**
well.. if this is too much effort I'd need the clear description where to find 'constants.h', then I would copy all the possible variables and write it in my own memory txt file..

revarbat, 2020-10-15:
These are all documented in the wiki at https://github.com/revarbat/BOSL/wiki/constants.scad

closed 2020-10-15 by revarbat

3Dietrich, 2020-10-15:
ha! How could i have missed that! However, I heard about BOSL2 and immersed myself in it