revarbat/BOSL #71: [BUG] Many warnings when using shapes.scad

kakila, 2020-09-15:
**Describe the bug**
Creating a tube produces all these wanrings

```
Used file cache size: 3 files
Compiling design (CSG Tree generation)...
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1211 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1211 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1211 
WARNING: undefined operation (undefined + undefined) in file shapes.scad, line 1211 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1211 
WARNING: undefined operation (undefined + undefined) in file shapes.scad, line 1211 
WARNING: undefined operation (undefined + undefined) in file shapes.scad, line 1211 
WARNING: undefined operation (number + undefined) in file shapes.scad, line 1211 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1212 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1212 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1212 
WARNING: undefined operation (undefined + undefined) in file shapes.scad, line 1212 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1212 
WARNING: undefined operation (undefined + undefined) in file shapes.scad, line 1212 
WARNING: undefined operation (undefined + undefined) in file shapes.scad, line 1212 
WARNING: undefined operation (number + undefined) in file shapes.scad, line 1212 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1213 
WARNING: undefined operation (number - undefined) in file shapes.scad, line 1213 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1213 
WARNING: undefined operation (undefined - undefined) in file shapes.scad, line 1213 
WARNING: undefined operation (undefined - undefined) in file shapes.scad, line 1213 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1213 
WARNING: undefined operation (undefined - undefined) in file shapes.scad, line 1213 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1214 
WARNING: undefined operation (number - undefined) in file shapes.scad, line 1214 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1214 
WARNING: undefined operation (undefined - undefined) in file shapes.scad, line 1214 
WARNING: undefined operation (undefined - undefined) in file shapes.scad, line 1214 
WARNING: undefined operation (undefined / number) in file shapes.scad, line 1214 
WARNING: undefined operation (undefined - undefined) in file shapes.scad, line 1214 
```
**Code To Reproduce Bug**
```
use <BOSL/shapes.scad>

tube(h=30, od=20, id=10);
```

**Expected behavior**
No warnings

**Additional context**
- OpenSCAD Version: 2020.09.12.nightly (git a86c176)

kakila, 2020-10-14:
Any news on this issue? This happens with the version in the repository for Ubuntu 20.04
```
https://download.opensuse.org/repositories/home:/t-paul/xUbuntu_20.04/
```
Using the 18.10 version doesn0t show the problem.

kakila, 2020-10-14:
The warings are caused by the use of first_defined with undefined operations in the arguments.
To avoid them one could do this:
```
    od1_2 = is_undef(od1)? undef : od1/2;
    d1_2 =  is_undef(d1)? undef : d1/2;
    od_2 =  is_undef(od)? undef : od/2;
    d_2 =  is_undef(d)? undef : d/2;
    ir1pw = is_undef(ir1)? undef : is_undef(wall)? undef : ir1+wall;
    id1_2pw = is_undef(id1)? undef : is_undef(wall)? undef : id1/2+wall;
    irpw = is_undef(ir)? undef : is_undef(wall)? undef : ir+wall;
    id_2pw = is_undef(id)? undef : is_undef(wall)? undef : id/2+wall;
    r1 = first_defined([or1, od1_2, r1, d1_2, or, od_2, r, d_2, ir1pw, id1_2pw, irpw, id_2pw]);
```
But seems wasteful. I hope you, with more experience in the language can make a better fix.

revarbat, 2020-10-15:
The OpenSCAD language has recently been changed to add these warnings in the dev releases.  Very annoying.  I fixed it in BOSL2, but it may take me a bit to fix it in BOSL1.

kakila, 2020-10-15:
I did not know about BOSL2, what is it? Is it a branch?

revarbat, 2020-10-15:
BOSL2 is a massive rewrite of BOSL1 that is NOT backwards-compatible with BOSL1. I've been working on it for a couple of years now. It's still under development and is not fully stable code yet. However, the major remaining changes left before release are bug fixes, documentation, or are related to the attachment code. 

https://github.com/revarbat/BOSL2

stefanct, 2020-10-25:
So when starting a new design (for the first time) with BOSL it is probably advisable to start with BOSL2 already? Asking for a friend ;)

revarbat, 2020-10-25:
I'm trying to bring BOSL2 to release quality, but I cannot yet guarantee I won't make minor API breaking changes.

commit 409d1e02 (2021-02-14): Check parameters for definition before doing any calculation (fix #71)

As of version 2021.01 OpenSCAD complains about operations applied to
undefined variables.
closed 2021-02-14 by revarbat