revarbat/BOSL #73: [BUG] - involute_gears are unusable

ftlpilot, 2020-09-29:
involute_gears. are unusable - the 2D shape is WRONG and doesn't extrude to 3D - this was supposedly fixed, but it' still there today - here's a screenshot of the example code.

[image]

This would be a really nice library - IF IT ACTUALLY WORKED !!!

mjparme, 2020-10-13:
I have used involute-gears just fine before. What do you mean it is unusable?

ftlpilot, 2020-10-13:
The inside is hollow and they can’t be rendered – look **CAREFULLY** at the pic I included, there is just an outer skin – THE CENTER IS MISSING, also this problem has been reported before.

 

Don’t rely on using them before – CHECK THEM NOW – THEY DON’T FUCKING WORK !!!!!!!!!!!!!

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

revarbat, 2020-10-15:
Took a bit to figure this one out. `gear2d()` was using tooth profiles that had very slightly overlapping lines between consecutive teeth.  When that path was converted to a polygon via `polygon()`, it *looked* fine, but internally had a bad geometry, so that `linear_extrude()` failed to extrude it correctly.

Fixed in 56756cc137281b4319dea5083262f22c7aa0b804

closed 2020-10-15 by revarbat

ftlpilot, 2020-10-15:
Thankx – much appreciated, I’ll try and work in a mention in my next DIYODE article about gears, probably Dec or Jan issue.

 

 

Thankx

Geoff Cohen

 

 <https://www.thingiverse.com/geoff_cohen/designs> https://www.thingiverse.com/geoff_cohen/designs 

 <https://diyodemag.com/projects/fixing_firmware_updating_firmware_on_the_arduino_bootloader_part_2> https://diyodemag.com/projects/fixing_firmware_updating_firmware_on_the_arduino_bootloader_part_2

 <https://diyodemag.com/education/exploring_3d_3d_printed_gears_part_1> https://diyodemag.com/education/exploring_3d_3d_printed_gears_part_1

 

https://www.avast.com/antivirus

ftlpilot, 2020-10-24:
Thanks again – gears work perfectly now – I’ll give your Library a mention in my next DIYODE article about Gears, in the next month or two.

 

FYI, the 1st Gears article is at  <https://diyodemag.com/education/exploring_3d_3d_printed_gears_part_1> https://diyodemag.com/education/exploring_3d_3d_printed_gears_part_1

 

Closed #73 <https://github.com/revarbat/BOSL/issues/73> .

https://www.avast.com/antivirus

revarbat, 2020-10-25:
Honestly, for gears code I'd use the involute_gears.scad file in the BOSL2 library, not BOSL1.   In BOSL2 I did a complete rewrite that renders faster and more accurately.  Also it has proper support for spiral bevel gears and worm gears.

ftlpilot, 2020-10-26:
Thankx – until you mentioned it I didn’t know there WAS a V2 – perhaps mentioning BOSL2 on https://www.openscad.org/libraries.html would be a good idea, at least when it’s out of alpha/beta testing

 

https://www.avast.com/antivirus

ftlpilot, 2020-11-18:
As you suggested, I tried the the involute_gears.scad file in the BOSL2 library – it’s pretty much unusable, firstly there was no include/use statement for attachments, when I inserted one in my code – this was the result

 

WARNING: Ignoring unknown function 'is_vector', in file ../../../../Users/Geoff/Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 220.

ERROR: Assertion 'is_vector(cp)' failed in file ../../../../Users/Geoff/Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 220

TRACE: called by 'attach_geom', in file ../../../../Users/Geoff/Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 910.

TRACE: called by 'attachable', in file ../../../../Users/Geoff/Documents/OpenSCAD/libraries/BOSL2/Involute_gears.scad, line 700.

TRACE: called by 'gear', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1590.

TRACE: called by 'translate', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1589.

TRACE: called by 'union', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1569.

TRACE: called by 'difference', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1568.

TRACE: called by 'Geoffs_double_helix', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1538.

TRACE: called by 'rotate', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1537.

TRACE: called by 'translate', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1536.

TRACE: called by 'color', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1535.

TRACE: called by 'union', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1521.

TRACE: called by 'difference', in file Geoffs-Dual-Geared-Extruder-All-3D-printed-Mk21v.scad, line 1520.

 

https://www.avast.com/antivirus

revarbat, 2020-11-18:
In BOSl2, you want to do:

```
include <BOSL2/std.scad>
include <BOSL2/involute_gears.scad>
```

ftlpilot, 2020-11-20:
Thankx again – missed the info about std.scad at the beginning of the file

 

https://www.avast.com/antivirus