revarbat/BOSL #78: [BUG]  Joiners are buggy in OpenScad 2019.05 (stable) as well as 2021.01 (AppImage)

andrei91ro, 2021-02-25:
**Describe the bug**
The joiners (<BOSL/joiners.scad>) do not render out correctly using their default parameter values.
On the 2019.04 OpenScad they sometimes contain one hole 

**Code To Reproduce Bug**
```
include <BOSL/constants.scad>
use <BOSL/joiners.scad>

translate([-6, -12.5, -50]) rotate([0, 90, 0]) half_joiner(screwsize=3, orient=ORIENT_X, align=V_ZERO);
translate([-6, 12.5, -50]) rotate([0, 90, 0]) half_joiner(screwsize=3, orient=ORIENT_X, align=V_ZERO);
```

**Expected behavior**
Add as many joiners as I want, and render them correctly.

By the way, congratulations for the library! I still love it!

**Screenshots**
The preview is correct (as shown in the image bellow) in both openscad versions.
In 2021.01 the rendering shows no objects at all:
[image]

For the 2019.05 version, the preview is again correct while the rendering (shown bellow) has a secondary hole. Also, if I try to import the model in a slicer such as PrusaSlicer, they do not appear at all.
[image]

**Additional context**
Add any other context about the problem here.

*Log for OpenSCAD 2019.05*
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 32
Geometry cache size in bytes: 189032
CGAL Polyhedrons in cache: 29
CGAL cache size in bytes: 1893184
Compiling design (CSG Products normalization)...
Normalized CSG tree has 2 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 32
Geometry cache size in bytes: 189032
CGAL Polyhedrons in cache: 29
CGAL cache size in bytes: 1893184
Total rendering time: 0 hours, 0 minutes, 0 seconds
Top level object is a 3D object:
Simple: no
Vertices: 180
Halfedges: 748
Edges: 374
Halffacets: 380
Facets: 190
Volumes: 1
UI-WARNING: Object may not be a valid 2-manifold and may need repair!
Rendering finished.

```
*Log for OpenSCAD 2021.01:*
```

Parsing design (AST generation)...
Saved backup file: /home/andrei/.local/share/OpenSCAD/backups/test_joiner-backup-jPkDCCjM.scad
Compiling design (CSG Tree generation)...
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 748 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 748 
Compiling design (CSG Products generation)...
Geometries in cache: 135
Geometry cache size in bytes: 437088
CGAL Polyhedrons in cache: 110
CGAL cache size in bytes: 13428256
Compiling design (CSG Products normalization)...
Normalized tree has 2 elements!
Compile and preview finished.
Total rendering time: 0:00:00.242
 
Parsing design (AST generation)...
Saved backup file: /home/andrei/.local/share/OpenSCAD/backups/test_joiner-backup-jPkDCCjM.scad
Compiling design (CSG Tree generation)...
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 748 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 783 
WARNING: undefined operation (undefined * vector) in file transforms.scad, line 748 
Rendering Polygon Mesh using CGAL...
Geometries in cache: 135
Geometry cache size in bytes: 437088
CGAL Polyhedrons in cache: 110
CGAL cache size in bytes: 13428256
Total rendering time: 0:00:00.255
WARNING: No top level geometry to render 
```

- OpenSCAD Version: 2019.05 (stable, Ubuntu 20.04) and 2021.01 (AppImage on Ubuntu 20.04)
- Other libraries used:
     No other library used

andrei91ro, 2021-02-25:
I forgot to mention that both OpenScad versions were using the most recent stable release of the BOSL library, v1.0.1

fe60, 2021-02-26:
This is fixed already but there is no official release containing commit a0bf8e9

andrei91ro, 2021-02-26:
I will test it out in the weekend and will close the issue afterwards.
Thanks for the info and for your work!

andrei91ro, 2021-02-26:
After testing on two machines, commit a0bf8e9 does indeed fix the issue.
I simply cloned the most recent version from git and everything was fine.
Thanks again!

closed 2021-02-26 by andrei91ro