revarbat/BOSL #86: [BUG] shapes.scad doesn't build because of syntax error in StaggeredSphere

watching-spd, 2021-04-30:
**Describe the bug**
shapes.scad won't build because of a syntax error if StaggeredSphere is uncommented:
```
ERROR: Parser error in line 1982: syntax error 
ERROR: Compilation failed! 
```

**Code To Reproduce Bug**
```
include <shapes.scad>;
```

**Expected behavior**
shapes.scad should build and import cleanly.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2015.03-2, Windows 10 (wow, this is a truly ancient version of OpenSCAD...)
- Other libraries used: None

fe60, 2021-04-30:
It works perfectly with OpenSCAD 2021.01 (and most probably does with 2019.05).
Is there a special reason for using such an old version of OpenSCAD?

watching-spd, 2021-05-01:
Nope! I've got a work-around (commenting out that module), but it might be nice to find the minimum supported version and throw a more useful error message in the case where the library uses newer language features?

fe60, 2021-05-02:
The error message is thrown by OpenSCAD. So there is nothing the library can do about it. And I'm not sure how OpenSCAD could handle this in a better way. It stumbles across an unknown statement and fails with "Syntax error". It can't know, it's a valid language feature in future versions.

I forked the git repository behind the wiki into a new one and started adding fixes. Its [README](https://github.com/fe60/bosl-wiki#readme) declares OpenSCAD 2019.05 as minimum version.
Once I think it is a feasible approach to host the documentation in a separate repository, I will ask the owner/author of BOSL to move the wiki into a dedicated Github-Repo to allow wiki contributions from everyone.

revarbat, 2021-05-16:
BOSL v1 wiki is now publicly editable.

revarbat, 2021-05-16:
I recoded `stagger_sphere()` to avoid using `each`.  It works with OpenSCAD 2015.03 now.

closed 2021-05-16 by revarbat